SpikeInterface/spikeinterface #3006: Only one cluster survives final clustering with KS4

EricKenjiLee, 2024-06-08:
@alejoe91 opening the issue here as per your instructions. SI version `0.100.6`; KS4 version `4.0`.

I've just started using SpikeInterface to try to compare DREDge to KS4 drift correction but I keep running into this bug in which all clusters but one are discarded during final clustering and then zero clusters make it through refractory period thresholding. This image is from after conducting drift correction and running Kilosort4 for spike detection and clustering using default parameters except `'do_correction' = False`. I've confirmed this error also occurs when using Kilosort's drift correction instead of DREDge and also that it happens on other recordings; when running sorting via KS4 GUI, I end up with hundreds of clusters to examine in Phy. See other screenshot showing it's a "typical" recording. I'm following the tutorial in the [docs](https://spikeinterface.readthedocs.io/en/latest/how_to/analyse_neuropixels.html).

Will upgrade KS4 to 4.0.11 and see if this resolves things.

[image]
[image]

JoeZiminski, 2024-06-10:
Oh I just saw you already did upgrade in #3007 😄 did it help?

EricKenjiLee, 2024-06-10:
Yes bumping to SI 0.100.7 and downgrading to kilosort4 version 4.0.6 worked! The only bug that remains is that this workflow takes approximately 5-7 times longer to get from raw data to spike clusters than using the KS GUI directly but I think your team is already aware of that issue. I'm working with a 8-core Intel i9 CPU, 64 GB of RAM, and a 4070 GPU. I'm also I/O bottlenecked as data is on an HDD but this is also the case for KS GUI. Will try to process again once my new NVMe SSD comes in.

closed 2024-06-13 by EricKenjiLee

EricKenjiLee, 2024-10-08:
Just commenting that I/O bottlenecking doesn't seem to have been the issue as speeds are still dreadfully slow even after loading data onto an NVMe SSD. GPU is being fully-utilized as well.