SpikeInterface/spikeinterface #3010: Cannot delete/load spike_locations when templates is missing

chrishalcrow, 2024-06-10:
The extension `spike_locations` relies on template information in its `init`. But spikeinterface initializes extensions when we delete them (when they're saved in a folder). So it's impossible to delete `spike_locations`, meaning it's impossible to do any recomputes on sorting analyzers with `spike_locations`.

Hence the following errors...

```
import spikeinterface.full as si
recording, sorting = si.generate_ground_truth_recording()
sorting_analyzer = si.create_sorting_analyzer(
    recording=recording, 
    sorting=sorting, 
    format="binary_folder", 
    folder="temp_folder",
    overwrite=True
)
sorting_analyzer.compute(["random_spikes", "templates", "spike_locations"])
sorting_analyzer.compute(["random_spikes"])
```

...since the recompute of `random_spikes` first deletes `templates`, then tries to delete `spike_locations`

The problem only occurs for the `spike_locations` extension.

zm711, 2024-06-12:
I wonder if this has to do with the patch for making sure qc metrics are calculated last? Since this is a child of templates I thought it would've gotten deleted before templates....

chrishalcrow, 2024-06-12:
> I wonder if this has to do with the patch for making sure qc metrics are calculated last? Since this is a child of templates I thought it would've gotten deleted before templates....

No, it's not about the deletion order. The reason is that currently, we calculate get_template_extremum_channel in the __init__ rather than in the compute (in _get_pipeline_nodes). We shouldn't do this (and don't do it for any other Extension) because we need the __init__ to run, even when we don't have the parents calculated.

PR #3015 (2024-06-21): Remove get_template_extremum_channel from spikelocations init
Fixes #3010

Removes `get_template_extremum_channel`, and hence the dependency on the `templates` extension, from the `SpikeLocations` init. It looks like we don't need to saves `spikes` at all in the extensions, I completely removed this line (this is similar to `ComputeSpikeAmplitudes`). @samuelgarcia can you double check this?

Note for #2984, on how to test this. The problem here was that `SpikeLocations` couldn't be initialised without `Templates` being present. To test this we could add something to the `AnalyzerExtensionCommo [... cut]
closed 2024-06-21 by samuelgarcia