SpikeInterface/spikeinterface #3012: sorting analyzer and analyzer extension core not installed through pip

SpencerBowles, 2024-06-11:
Hello, I was setting up spike interface on a new computer, and when I pip installed spike interface in my conda environment, the sortinganalyzer.py and analyzer_extension_core.py were not included in the installation. Are these no longer supported? They are still in the docs and API. 

If I pip install using the github HTTP they install fine, though I now get warnings about potential version issues.

I tried `pip install spikeinterface` `pip install spikeinterface[full]` and `pip install spikeinterface[full,widgets]` and they all lack those files. I am installing version 0.100.7

alejoe91, 2024-06-11:
Hi @SpencerBowles 

The sorting analyzer is not officially released, but you can install  the `0.101.0rc0` pre-release with:
```
pip install --pre spikeinterface[full,widgets]
```

SpencerBowles, 2024-06-11:
Awesome, thank you!

closed 2024-06-11 by alejoe91