SpikeInterface/spikeinterface #3018: Error in Kilosort When Converting intan .rhs Data

Fusenheizi, 2024-06-12:
Hello,

I encountered an issue when converting intan .rhs data in Kilosort. 

If I do not specify the stream_id or stream_name, the following error occurs:

[image]

’‘’
Traceback (most recent call last):

  File "C:\Users\319_exp\.conda\envs\kilosort\lib\site-packages\kilosort\gui\converter.py", line 265, in convert_to_binary
    
recording = get_recording_extractor(

  File "C:\Users\319_exp\.conda\envs\kilosort\lib\site-packages\kilosort\gui\converter.py", line 308, in get_recording_extractor
    
return getattr(extractors, extractor_name)(filename, **kwargs)

  File "C:\Users\319_exp\.conda\envs\kilosort\lib\site-packages\spikeinterface\extractors\neoextractors\intan.py", line 34, in __init__
    
NeoBaseRecordingExtractor.__init__(

  File "C:\Users\319_exp\.conda\envs\kilosort\lib\site-packages\spikeinterface\extractors\neoextractors\neobaseextractor.py", line 207, in __init__

raise ValueError(

ValueError
: 
This reader has several streams: 
Names: ['RHD2000 amplifier channel', '']
IDs: ['0', '11']. Specify it with the 'stream_name' or 'stream_id' arguments
‘’‘

Initially, I thought the stream_name might be related to my recording ports. However, after testing with .rhs files containing different numbers of ports, I found that the error asking to specify ['0', '11'] persists regardless of the number of ports. **Therefore, I believe stream_id or stream_name might not be linked to the number of ports, but I am unsure what they are associated with.**

I have attached my test data (size: 300MB, I use wormhole to upload), which includes data from four ports of the Intan RHS system, totaling 128 channels. Each port is connected to one rat, with 32 channels per rat.

**Datalink**: https://wormhole.app/14jK4#tTtXxo1cfAUaIigeI4nPRQ

Additionally, I tried specifying stream_id and stream_name as 0 or 11 and then converting to .bin format. However, the converted data does not display in the Kilosort interface as expected, regardless of the number of ports in the Intan data.

[image]

Could you please provide some guidance on how to resolve this issue?

Thank you! ✨

alejoe91, 2024-06-12:
Hi @Fusenheizi 

Can you point us to the code for the GUI you are using for conversion? It was not written by us ;)

Fusenheizi, 2024-06-12:
> Hi @Fusenheizi
> 
> Can you point us to the code for the GUI you are using for conversion? It was not written by us ;)

Yeah, I see. (ノへ￣、) This is the code for the data converter GUI in Kilosort. I'm not sure if it will be helpful.

https://github.com/MouseLand/Kilosort/blob/main/kilosort/gui/converter.py

zm711, 2024-06-12:
RHS has three data streams all related to the amplifier with stim data and dc amplifier as well. `stream_id='0'` refers specifically to the amplifier data whereas I 11 is a different stream. If they are using Neo at  base you could also open the issue over on Neo and we can see if there is a stream_id -> name issue but I'm assuming for your purposes you want `stream_id='0'`.

h-mayorquin, 2024-06-13:
The data is not available.

Fusenheizi, 2024-06-14:
> RHS has three data streams all related to the amplifier with stim data and dc amplifier as well. `stream_id='0'` refers specifically to the amplifier data whereas I 11 is a different stream. If they [...]

Thank you! I understand it now. I do need the data for stream_id='0'. Unfortunately, even with stream_id='0', Kilosort still shows that the data is empty. I think this might be an issue for the authors of Kilosort to resolve.

Fusenheizi, 2024-06-14:
> The data is not available.

Sorry(；′⌒`), since the link from Wormhole is only valid for 24 hours, I have re-uploaded it. This link should be usable.
https://wormhole.app/aWnkM#xeV8wzvKtRc_h23kcCDqBg

h-mayorquin, 2024-06-14:
Hi, yes, your data looks like this and loads correctly:

[image]

It seems fine to me but you can confirm it with the script in the screenshoot.

I am closing this as this is mainly an issue with Kilosort, they need to specify the amplifier stream for Intan. If you have some feedback that throws the ball back here feel free to open another issue.

closed 2024-06-14 by h-mayorquin