SpikeInterface/spikeinterface #3019: Discussion on comment headers / sections to help split up the code

JoeZiminski, 2024-06-12:
For longer modules it is common to add comment sections to help readability. For example in SpikeInterface you might have:

```
# LOW-LEVEL IMPLEMENTATIONS
def compute_correlograms_numpy(sorting, window_size, bin_size):
```

In the SpikeInterface convention (as above) is a comment above the function that starts the section. Sometimes I find these a little hard to see where sections start and end and was wondering if there is any appetite for using a different convention (some examples below).

Personally I find such headers make navigating the code easier but I think they can be a little contentious and I don't have particularly strong feelings either way, but thought I'd start a discussion!

```
# ------------------------------------------------------
# Low Level Implementations
# ------------------------------------------------------
```

```
# Low Level Implementations
# ------------------------------------------------------
```

```
#########################################
# Low Level Implementations
#########################################
```

alejoe91, 2024-06-12:
Thanks for bringing this up @JoeZiminski 

I think we should go with better headers for sure! Personally, I like this:
```
#########################################
# Low Level Implementations
#########################################
```

h-mayorquin, 2024-06-12:
While this is an improvement with very low cost. If many of those are needed is probably indicative that a new file / module is required instead. That is, the module contains probably too many things if this occurs.

The (really small!) cost is:
* Having to thing on where to put your function (maybe not bad) and the arguments that arise from it (coordination cost).
* Small maintenance cost to keep them tidy.

JoeZiminski, 2024-06-13:
Thanks both! I agree @h-mayorquin that a good end goal is to not have these at all. I think short-term these can be useful as indicators of where things could be factored out, and simplify future refactorings?

h-mayorquin, 2024-06-13:
Yes, that makes sense to me.

samuelgarcia, 2024-06-14:
I prefer sober stuff
```
# Low Level Implementations
# -------------------------
```

JoeZiminski, 2024-06-14:
I don't have a strong feeling but have a preference for lighter edges that to me attract attention less. Because we have another first-past-the-post election coming up and I am jealous of your European PR system I will propose a PR style vote (I have numbered above) with [alternative vote rules](https://www.electoral-reform.org.uk/voting-systems/types-of-voting-system/alternative-vote/).

JoeZiminski, 2024-06-14:
I don't have a strong feeling but have a preference for lighter edges that to me attract attention less. Because in the UK have another first-past-the-post election coming up and I am jealous of your European PR system I will propose a PR style vote (I have numbered above) with [alternative vote rules](https://www.electoral-reform.org.uk/voting-systems/types-of-voting-system/alternative-vote/). My vote in order of preference would be [1, 2, 3]. However, if people don't agree this is a good approach we can have a vote on the best approach to decide 😆

zm711, 2024-06-18:
This what Sam did in [Neo](https://github.com/NeuralEnsemble/python-neo/blob/d8639b41675d135649eb6fc9418d30db261c33d2/neo/rawio/intanrawio.py#L776-L777). I like the hashes because they provide a nicer contrast. 

```
##########################
# Low Level Implementation
```

So I would choose this style
followed by 3, 1, 2 #3016 
:)

chrishalcrow, 2024-06-19:
I also vote 1, 2, 3. There's something about hashtags that hurt my eyes.

alejoe91, 2024-06-24:
I vote for 3, 1, 2, @h-mayorquin @yger your votes are needed for democracy!

h-mayorquin, 2024-06-24:
Let's keep the matlab tradition 3, 1, 2 as well.

JoeZiminski, 2024-07-18:
@yger do you have any opinions on this critical topic 😄?

yger, 2024-07-18:
I vote for 3,1,2 !

h-mayorquin, 2024-09-12:
Let's add this to the dev documentation and consider this issue close. Decisions are tiresome and we can implement it piece wise. 

Thanks for leading this @JoeZiminski .

h-mayorquin, 2024-09-12:
Here:
https://github.com/SpikeInterface/spikeinterface/pull/3402

PR #3402 (2024-09-12): Add decision about headers to dev docs
Should close #3019.