SpikeInterface/spikeinterface #3020: bug in '0.101.0rc0'

shahafweissMPI, 2024-06-12:
when trying to run kilosort4 i get an error that more values were returned by "get_run_parameters(ops)" than expected in kilosort4.py.
this was fixed by adding 2x "_" to the output variables.

spikeinterface.__version__
Out[2]: '0.101.0rc0'

ss.Kilosort4Sorter.get_sorter_version()
Out[4]: '4.0.11'

File d:\dropbox\cambridge\code\preprocess2.py:134 in run_sorter
    sorting = si.run_sorter_local(sorter_name=sorter, recording=rec, folder=out_folder ,remove_existing_folder=True, **params)

  File D:\GitHub\NewGit\spikeinterface24\src\spikeinterface\sorters\runsorter.py:245 in run_sorter_local
    SorterClass.run_from_folder(folder, raise_error, verbose)

  File D:\GitHub\NewGit\spikeinterface24\src\spikeinterface\sorters\basesorter.py:292 in run_from_folder
    raise SpikeSortingError(

SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "D:\GitHub\NewGit\spikeinterface24\src\spikeinterface\sorters\basesorter.py", line 257, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "D:\GitHub\NewGit\spikeinterface24\src\spikeinterface\sorters\external\kilosort4.py", line 199, in _run_from_folder
    n_chan_bin, fs, NT, nt, twav_min, chan_map, dtype, do_CAR, invert, _, _, tmin, tmax, artifact = (
ValueError: too many values to unpack (expected 14)

Spike sorting failed. You can inspect the runtime trace in E:\SI_out_test\scratch\spk_out\afm16963\afm16963_240525_home_cage_g0\afm16963_240525_home_cage_g0_imec0\kilosort4/spikeinterface_log.json.

the fix:

        n_chan_bin, fs, NT, nt, twav_min, chan_map, dtype, do_CAR, invert, _, _, tmin, tmax, artifact,_,_ = (
            get_run_parameters(ops)
        )

alejoe91, 2024-06-12:
Hi @shahafweissMPI 

We still need to patch SI to support the latest kilosort versions. Last supported version is 4.0.6

alejoe91, 2024-06-24:
@shahafweissMPI this is now fixed both in version 0.100.8 and in main.

A new docker image for v4.0.12 is available on docker hub: https://hub.docker.com/layers/spikeinterface/kilosort4-base/4.0.6_cuda-12.0.0/images/sha256-e6c8919a949d96460fd52b91ba092c2d498984670eef548a79e49844d35da0cf?context=explore

closed 2024-06-24 by alejoe91

WeissShahaf, 2024-06-25:
thanks @alejoe91 !
i'm assuming the 4.0.12 version is at: https://hub.docker.com/layers/spikeinterface/kilosort4-base/4.0.12_cuda-12.0.0/images/sha256-e6c8919a949d96460fd52b91ba092c2d498984670eef548a79e49844d35da0cf?context=explore right?

for non-docker it just works on main>=0.1008 ?

alejoe91, 2024-06-25:
It works both on main and 0.100.8 ;)