SpikeInterface/spikeinterface #3026: Question about Pipeline Design：postprocessing part

Alchemist-Y, 2024-06-13:
Question: Why the poseprocessing part is behind the manual curation?
Situation: Actually, after the manual curation with Phy, such as merging the clusters or split them, which do will influence the mean waveform parameters. However, the designed pipeline will define the waveform extractor for the basesorter and the baserecording object, which means  the impregnability state from the manual curation. I think it is a little hard to understand. 
Maybe I got some misunderstanding of it. Hope to get your professional advice!

alejoe91, 2024-06-13:
Hi @Alchemist-Y 

The postprocessing is needed in order to produce all additional data (e.g., spike amplitudes, PCA scores, etc.) that are used to export to Phy

Alchemist-Y, 2024-06-13:
hi, @alejoe91 
I konw this aim, but after the manual curation with Phy, some attribute with the well isolated unit do will be changed, then the postprocessing part especially the waveform or firing rate will change, that means the precomputed parameters won't give the accurate information for the unit.

alejoe91, 2024-06-13:
Yes you should recompute the postprocessing after manual curation. We are working on a layer that allows to automatically re-compute postprocessed data after merges/splits, but it's WIP

Alchemist-Y, 2024-06-13:
OK, Thanks a lot. And I will also try some ideas to solve the feedback pipeline! And really waiting for your update!

closed 2024-06-13 by Alchemist-Y