SpikeInterface/spikeinterface #3027: Import style in code-base

alejoe91, 2024-06-13:
Hi guys,

I think we should unify the way we import internal functions across the code-base.

Currently we have a lot of relative imports + some full imports in the source code. Od course tests should keep full imports.

All in favor?

@samuelgarcia @h-mayorquin @zm711 @chrishalcrow @JoeZiminski @yger

JoeZiminski, 2024-06-13:
Agreed! I was swayed to use full imports in another package by [this top answer](https://stackoverflow.com/questions/4209641/absolute-vs-explicit-relative-import-of-python-module). This can also be linted in [ruff](https://docs.astral.sh/ruff/rules/relative-imports/) for enforcement if desired.

alejoe91, 2024-06-13:
Interesting! yeah maybe then we should just switch to absolute imports. Anyone has strong feelings about this?

h-mayorquin, 2024-06-13:
@alejoe91 are you proposing to make all imports full imports in the code base?

I personally prefer absolute imports as well. We use relative imports in neuroconv and I often find myself wondering what is the location of some something and that is occluded by the dots (e.g. I don't know where `..a_module` is).

The only advantage that I see of relative imports is that module re-organization might be easier but my IDE takes care of that problem for me if it ever arises (and it should arise very seldom!).

alejoe91, 2024-06-13:
I'm proposing to unify imports! If you guys think that absolute are better, let's go for it!

h-mayorquin, 2024-06-13:
Ah, got you, thanks, yeah, I prefer absolute imports (and that would be less work, I feel, right?).

samuelgarcia, 2024-06-14:
I prefer relative :)
of course.

zm711, 2024-06-17:
Just got back from vacation :) 

I think from bring in new people perspective absolute is the way to go. It gives provenance (which we all love). I like the look of relative better, but I think we want to be as inclusive as possible for people so new folks can easily navigate the code base :)

zm711, 2025-02-05:
Well are we good on this or is there more debate? Basically we all agreed to absolute except Sam and then we never did anything :) haha.

chrishalcrow, 2025-02-06:
Now that I've been coding a bit longer, I can vote! Also pro absolute imports, and am using them in other projects I'm working on.

chrishalcrow, 2025-03-13:
Done in #3766