SpikeInterface/spikeinterface #3028: Discussion on best way to visualise data in an experiment

JoeZiminski, 2024-06-13:
At the hackathon, there was a lot of discussion on the importance of data visualisation. There are already a lot of nice tools for visualising data in spikeinterface (`plot_traces()`, and [spikeinterface-gui](https://github.com/SpikeInterface/spikeinterface-gui)). These are very useful for having a good look, close up at a single dataset. 

However, when running a lot of acquisition sessions, it can be useful to print summary of a recording to quickly assess data quality across an entire experiment. I was wondering how people are handling this in their research? I sometimes use code blocks like the below (psedocode) to dump images to a PDF for review later. An intrinsic problem at what scale to plot the data - at large time bins you loose all spike resolution. At small time bins, you need to take small snapshots through a long recording. In this case it might be easy to miss large blocks of the recording where data quality could feasibly be comprimised. Also,  if saving to `.svg / .png` you cant really zoom in like a usual matplotlib recording. 

<details open>
<summary>Want to ruin the surprise?</summary>
<br>
    ````
    from matplotlib.backends.backend_pdf import PdfPages
    import spikeinterface.widgets as si_widgets
    
    pdf_pages = PdfPages(output_filepath)
    fig, ax = plt.subplots(*(1, 3))

    # some list of times to plot, with a period (i.e. window) to plot

    for i, (start_time, stop_time) in enumerate(
            zip(times,
                times + period_s)
     ):

            si_widgets.plot_traces(
                recording,
                order_channel_by_depth=True,
                time_range=(start_time, stop_time),
                mode="line",
                show_channel_ids=True,
                return_scaled=True,
                ax=ax[i],
            )
            ax[i].get_legend().remove()

    pdf_pages.savefig(fig)
    plt.close(fig)
    pdf_pages.close()
    ````
</details>

samuelgarcia, 2024-06-14:
Hi Joe.
I am also using the `export_report(analyzer)` which export many figure into a folder.
We could also have `export_several_report(list_of_analyzer)` that would generate less figures for a more global overview on sortings.
This export report is quite basic at the moment but thsi is more or less align whith what you are mentionning no ?

JoeZiminski, 2024-06-14:
Thanks Sam, this is great I forgot about `export_report()` which is very useful for this purpose. But I also mean to visualise the raw traces before / after certain preprocessing steps, just to check a) the raw data (well, after filtering) looks as expected across the recording b) all preprocessing steps are not having an unintended consequences

JoeZiminski, 2024-06-21:
Thanks @zm711, the scope of this was more focused on getting an overview across a whole experiment relatively quickly, in a way that maximises the information, it would be good to get your thoughts on this. For example say you have 3x 1h sessions per day, 10 sessions per animals, 25 animals. I think the most convenient way is to use something like `export_report` with some trace snippets to get an overview of certain steps (also, for example, alongside the drift-correction report). Spikeinterface-gui is great for getting a close-up view of  single session but might be less efficient for a reporducable, whole-experiment overview. 

A part of the question is - how many time points and at what scale is it necessary to save trace snippets in order to maximise the information? For example, you can plot the entire 1h session, but the resolution will be too poor to see anything more than major drift. You could 60 1s snippets once a minute, but you will be drowning in images. Maybe a solution is to add functionality to export-report or something similar with options to select exact time points to plot or the windowing strategery for creation of trace snippet plots.