SpikeInterface/spikeinterface #3039: Spike sorting Error : attribute error due to `np.issctype' removed from Numpy 2.0

taningh86, 2024-06-18:
Hi,
This issue may be relevant to 'Support Numpy 2.0' issue opened by @h-mayorquin yesterday. Looks like `np.issctype' was removed from Numpy 2.0 which is resulting in a spikesorting error. It says 'AttributeError: `np.issctype` was removed in the NumPy 2.0 release. Use `issubclass(rep, np.generic)` instead'. I tried to change it to 'issubclass(obj, np.generic)' but surprisingly the 'core' folder in spikeinterface folder is not there, therefore, no core_tools.py' file where I could make these changes. It surprised me because these files/folder are there in my personal laptop but cannot be found in the computer where I actually run analysis. The spikeinterface folder in the analysis computer only has two folders: postprocessing and qualitymetrics. When I updated SI some months ago I pulled it from github and I believe its doing automatic updates anytime there's an update or changes. However, if that's the case, it is still surprising that there are no other folder besides these two.
Anyways, I will appreciate if you guys can guide me how to resolve this.

Thanks
J

zm711, 2024-06-18:
@taningh86, 

I'm not sure what this folder problem is, which seems confusing. I might need a screen shot of the folder you're looking at?

As far as NumPy 2.0 it isn't supported yet. So I would downgrade NumPy for now so you don't run into these issues. We are working on that.

h-mayorquin, 2024-06-18:
@taningh86 so if I understand correctly your local version you downloaded from the repo directly and you periodically run fetch and pull? 

To modify the files when you install that way you need to do an editable install https://setuptools.pypa.io/en/latest/userguide/development_mode.html

The simplest thing is to fetch the latest version though as we have already merged those changes. Just re-install spikeinterface from the latest version on main.

taningh86, 2024-06-18:
@h-mayorquin Yeah, I downloaded it directly from the repo and I would assume its doing fetch and pull by itself since I don't update SI on my own thinking that downloading directly from the repo would do updates automatically. 
Also, this spikesorting error involving Numpy 2.0 is a new thing and KS2.5 worked fine without issues until few days ago. I will re-install SI and see if that resolves the issue.
@zm711 The folder is not the issue but the spikesorting error which says - ```AttributeError: np.issctype was removed in the NumPy 2.0 release. Use `issubclass(rep, np.generic)` instead```. The folders were something  I observed in the anaconda3>envs>si_env>lib>site-packages>spikeinterface folder. I am seeing only 3 folders (sorry not 2) - curation, postprocessing, and qualitymetrics folder in the computer where I actually run the analysis. Image below:
[image]

Isn't there supposed to be lot more folders like how it is in this image below? The one below is from my personal laptop. 
[image]
 In my personal computer I did not install SI directly from the repo. So I was just wondering if the spikesorting error involving Numpy 2.0 I am seeing above in my analysis computer is due to a recent update in spikeinterface, since I downloaded it directly from the repo there. Hope this makes things clear. let me know

zm711, 2024-06-18:
It's weird though because typically things don't auto-update. The only way you'd get NumPy 2.0 is if you downloaded it into your environment within the past day or two (I haven't checked the exact release date). So if you hadn't updated the environment then it shouldn't be there.  How does your analysis computer get updated? 

As @h-mayorquin said he patched some of the main errors so fetching the current spikeinterface should fix that issue. But maybe there are symlinks between the other folders. Or do you get an error if you do

```python
import spikeinterface.extractors as se
```

I'm less focused on the NumPy error since we can fix that with the install. I'm more curious about the weird sub-set of folders.

h-mayorquin, 2024-06-18:
> @h-mayorquin Yeah, I downloaded it directly from the repo and I would assume its doing fetch and pull by itself since I don't update SI on my own thinking that downloading directly from the repo would do updates automatically.

No, as @zm711 mentions, this is not the case by default. You need to fetch and merge (or pull) from time to time. 

And yes, your folders should look like the one below. Something weird is going on. Can you re-install and see if you get the same structure?

taningh86, 2024-06-18:
Even after re-installing SI I am still getting the same error and only 3 folders. Looks like there's an issue here. FYI I uninstalled spikeinterface and reinstalled directly from the source again ( i did not do pip install spikeinterface[full,widgets]. I am still getting the same error. Below is the error in more detail :
```SpikeSortingError: Spike sorting in docker failed with the following error:
Traceback (most recent call last):
  File "/Cat_GT_Out/catgt_NPX2_New_mouse_6_12_24_2LHA_2RSP_New_cagewall_real_FOR_g0/in_container_sorter_script.py", line 23, in <module>
    sorting = run_sorter_local(
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py", line 222, in run_sorter_local
    output_folder = SorterClass.initialize_folder(recording, output_folder, verbose, remove_existing_folder)
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py", line 145, in initialize_folder
    recording.dump(rec_file)
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/core/base.py", line 635, in dump
    self.dump_to_json(file_path, relative_to=relative_to, folder_metadata=folder_metadata)
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/core/base.py", line 678, in dump_to_json
    json.dumps(dump_dict, indent=4, cls=SIJsonEncoder),
  File "/home/miniconda3/lib/python3.9/json/__init__.py", line 234, in dumps
    return cls(
  File "/home/miniconda3/lib/python3.9/json/encoder.py", line 201, in encode
    chunks = list(chunks)
  File "/home/miniconda3/lib/python3.9/json/encoder.py", line 431, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/home/miniconda3/lib/python3.9/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/home/miniconda3/lib/python3.9/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/home/miniconda3/lib/python3.9/json/encoder.py", line 438, in _iterencode
    o = _default(o)
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/core/core_tools.py", line 95, in default
    if np.issctype(obj):  # Cast numpy datatypes to their names
  File "/root/.local/lib/python3.9/site-packages/numpy/__init__.py", line 397, in __getattr__
    raise AttributeError(
AttributeError: `np.issctype` was removed in the NumPy 2.0 release. Use `issubclass(rep, np.generic)` instead.```

And, No I don't get any error when I do `import spikeinterface.extractors as se`
Also, I did not update the environment but I think the analysis computer gets updated on its own automatically. That seems to be out of my control since I have set it to stop automatic updates but it will still do updates on its own. Would that be an issue? If that is so then it has automaic updates several times over the months but did not cause any issue until today. 
The folder issue is troubling. Let me know what yo guys need to diagnose it. 
Thanks!!

taningh86, 2024-06-18:
Sorry the last paragraph got into the code block.

h-mayorquin, 2024-06-18:
I would need you to describe more in detail how you are installing spikeinterface. Can you post the set of instructions you use so I can run them on my side? Are you using windows, ubuntu? Also, can you do it in a clean environment just to be sure.

taningh86, 2024-06-19:
For sure! What I did today is below:
1. I just re-installed SI  from the source in anaconda. First I did `pip uninstall spikeinterface`.
2. Then I pasted `git clone https://github.com/SpikeInterface/spikeinterface.git
cd spikeinterface
pip install -e .
cd ..`
3. It installed sucessfully.

I could have just done `pip install spikeinterface[full,widgets]` but it installs 100.7 version not 101.0. So I had to install from the source. 
I will delete the environment and make a fresh one and see if that changes anything. Will update you in a few.

h-mayorquin, 2024-06-19:
Thanks. Hope we can figure it out.

atombysx, 2024-06-19:
Hi all,

I'm encountering the same problem trying to run kilosort3:
`probe0_sorting_ks3 = si.run_sorter(sorter_name= 'kilosort3',recording=probe0_preprocessed_corrected,output_folder=save_folder+'probe0/sorters/kilosort3/',docker_image=True,do_correction=False)`

The error is the following:
SpikeSortingError: Spike sorting in docker failed with the following error:
Traceback (most recent call last):
  File "/home/lab/spikeinterface_sorting/temp_data/20240605/probe0/sorters/in_container_sorter_script.py", line 23, in <module>
    sorting = run_sorter_local(
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py", line 222, in run_sorter_local
    output_folder = SorterClass.initialize_folder(recording, output_folder, verbose, remove_existing_folder)
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py", line 145, in initialize_folder
    recording.dump(rec_file)
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/core/base.py", line 635, in dump
    self.dump_to_json(file_path, relative_to=relative_to, folder_metadata=folder_metadata)
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/core/base.py", line 678, in dump_to_json
    json.dumps(dump_dict, indent=4, cls=SIJsonEncoder),
  File "/home/miniconda3/lib/python3.9/json/__init__.py", line 234, in dumps
    return cls(
  File "/home/miniconda3/lib/python3.9/json/encoder.py", line 201, in encode
    chunks = list(chunks)
  File "/home/miniconda3/lib/python3.9/json/encoder.py", line 431, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/home/miniconda3/lib/python3.9/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/home/miniconda3/lib/python3.9/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/home/miniconda3/lib/python3.9/json/encoder.py", line 438, in _iterencode
    o = _default(o)
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/core/core_tools.py", line 96, in default
    if np.issctype(obj):  # Cast numpy datatypes to their names
  File "/root/.local/lib/python3.9/site-packages/numpy/__init__.py", line 397, in __getattr__
    raise AttributeError(
AttributeError: `np.issctype` was removed in the NumPy 2.0 release. Use `issubclass(rep, np.generic)` instead.

I first encountered this when I was trying out the new 101 version.

But I removed the conda environment completely and re-installed a new environment with 100.7 stable version. But still the same popped up which never happened to me before.

`pip show numpy` also shows that my numpy version is 1.26.4

alejoe91, 2024-06-19:
@atombysx the problem in your case is that we added the upper version bound for numpy on main, but v0.100.7 (which gets installed in the docker container), doesn't have this yet. 

Unfortunately this will need to wait until the next release. @h-mayorquin @zm711 maybe worth making a 0.100.8 with the numpy upper bound and related fixes?

h-mayorquin, 2024-06-19:
That makes sense to me. Can't user installs the latest main inside the docker as well to circumvent this problem?

zm711, 2024-06-19:
I think a point 0.100.8 is a great idea. That set of code likely will never be 2.0 compatible so we should try to protect people that want to stay on 0.100.x.

taningh86, 2024-06-20:
Have we found a solution to this yet. should I downgrade to 100.8? I have been using 101.0 version and would like to keep that. However if there's no way around it now then I can install 0.100.8. Please let me know.

h-mayorquin, 2024-06-20:
To summarize there are three options:
1) If you are well versed in docker you can do something along the lines of what Alessio says.
2) See if @alejoe91 decides to do a new release.
3) Downgrade for a while, we will have a release in a month at most.

I vote for making a new release with the latest kilosort changes and the numpy bound.

taningh86, 2024-06-20:
@h-mayorquin I will downgrade for now and just wait for the new release. Thanks for everyone's input!

AOONIH, 2024-06-21:
Hi. I'm also running in to the same issue and I am still using version 0.100.1. Is there a reason why the docker image is installing numpy 2.0?

zm711, 2024-06-21:
@AOONIH 

Yeah, the docker is trying to install from the 100.x pip release which doesn't have the numpy limit in its pyproject.toml so it's trying to grab the most uptodate numpy. This is fixed on main, but we need to prep a release on the 100.x (see #3060) for this to work with the pip installable version.

taningh86, 2024-06-21:
I removed the si_env environment, reinstalled SI using `pip install spikeinterface[full,widgets]`, created a fresh si_env with `conda env create --file full_spikeinterface_environment_windows.yml`. The version is 100.7. Now when I run the sorter in docker I get the module not found error
```
sorter_params = job_kwargs
sorting_KS25 = ss.run_sorter('kilosort2_5', recording_saved, 
                            output_folder=base_folder / 'results_KS25',
                            verbose=True, docker_image=True, 
                            **sorter_params)
```
Here's the error
```
ModuleNotFoundError                       Traceback (most recent call last)
Cell In[6], line 2
      1 sorter_params = job_kwargs
----> 2 sorting_KS25 = ss.run_sorter('kilosort2_5', recording_saved, 
      3                             output_folder=base_folder / 'results_KS25',
      4                             verbose=True, docker_image=True, 
      5                             **sorter_params)

File ~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py:169, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, delete_container_files, with_output, **sorter_params)
    167         else:
    168             container_image = singularity_image
--> 169     return run_sorter_container(
    170         container_image=container_image,
    171         mode=mode,
    172         **common_kwargs,
    173     )
    175 return run_sorter_local(**common_kwargs)

File ~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py:457, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, delete_container_files, extra_requirements, installation_mode, spikeinterface_version, spikeinterface_folder_source, **sorter_params)
    454     py_user_base_folder.mkdir(parents=True, exist_ok=True)
    455     py_user_base_unix = path_to_unix(py_user_base_folder)
--> 457 container_client = ContainerClient(mode, container_image, volumes, py_user_base_unix, extra_kwargs)
    458 if verbose:
    459     print("Starting container")

File ~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\container_tools.py:87, in ContainerClient.__init__(self, mode, container_image, volumes, py_user_base, extra_kwargs)
     85 if mode == "docker":
     86     if not HAS_DOCKER:
---> 87         raise ModuleNotFoundError("No module named 'docker'")
     88     client = docker.from_env()
     89     if container_requires_gpu is not None:

ModuleNotFoundError: No module named 'docker'
```
Not sure why is this. I don't think it is an issue with using a different python interpreter other than the si_env conda environment. However, I do have multiple python interpreters in the computer (which is pretty common) and this was never an error I remember getting so far since I started using SI. Any suggestions?

alejoe91, 2024-06-21:
You just need to `pip install docker`

h-mayorquin, 2024-06-21:
Note that at some point me and @samuelgarcia argued against @alejoe91 use of the installation_msg strings claiming that it would be obvious what to do for the user when they get one of those errors

This is evidence against our position.

I do think that this probably should be fix at the std lib level though. I will open an issue in python discussion when I have some time.

taningh86, 2024-06-21:
@alejoe91 I indeed did pip install docker

taningh86, 2024-06-21:
It does not work. Should I uninstall docker too and reinstall it?

alejoe91, 2024-06-21:
Same error?

alejoe91, 2024-06-21:
What's the output of `pip list | grep docker`?

taningh86, 2024-06-21:
ok. I just had to restart the kernel and it is not giving the 'module not found error'. However, It is still giving the initial error that I started this issue thread for even after downgrading to 100.7. 
```
SpikeSortingError                         Traceback (most recent call last)
Cell In[5], line 2
      1 sorter_params = job_kwargs
----> 2 sorting_KS25 = ss.run_sorter('kilosort2_5', recording_saved, 
      3                             output_folder=base_folder / 'results_KS25',
      4                             verbose=True, docker_image=True, 
      5                             **sorter_params)

File ~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py:169, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, delete_container_files, with_output, **sorter_params)
    167         else:
    168             container_image = singularity_image
--> 169     return run_sorter_container(
    170         container_image=container_image,
    171         mode=mode,
    172         **common_kwargs,
    173     )
    175 return run_sorter_local(**common_kwargs)

File ~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py:595, in run_sorter_container(sorter_name, recording, mode, container_image, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, delete_container_files, extra_requirements, installation_mode, spikeinterface_version, spikeinterface_folder_source, **sorter_params)
    593 if run_error:
    594     if raise_error:
--> 595         raise SpikeSortingError(f"Spike sorting in {mode} failed with the following error:\n{run_sorter_output}")
    596 else:
    597     if with_output:

SpikeSortingError: Spike sorting in docker failed with the following error:
Traceback (most recent call last):
  File "/Cat_GT_Out/catgt_NPX2_New_mouse_6_12_24_2LHA_2RSP_New_cagewall_real_FOR_g0/in_container_sorter_script.py", line 23, in <module>
    sorting = run_sorter_local(
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/runsorter.py", line 222, in run_sorter_local
    output_folder = SorterClass.initialize_folder(recording, output_folder, verbose, remove_existing_folder)
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/sorters/basesorter.py", line 145, in initialize_folder
    recording.dump(rec_file)
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/core/base.py", line 635, in dump
    self.dump_to_json(file_path, relative_to=relative_to, folder_metadata=folder_metadata)
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/core/base.py", line 678, in dump_to_json
    json.dumps(dump_dict, indent=4, cls=SIJsonEncoder),
  File "/home/miniconda3/lib/python3.9/json/__init__.py", line 234, in dumps
    return cls(
  File "/home/miniconda3/lib/python3.9/json/encoder.py", line 201, in encode
    chunks = list(chunks)
  File "/home/miniconda3/lib/python3.9/json/encoder.py", line 431, in _iterencode
    yield from _iterencode_dict(o, _current_indent_level)
  File "/home/miniconda3/lib/python3.9/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/home/miniconda3/lib/python3.9/json/encoder.py", line 405, in _iterencode_dict
    yield from chunks
  File "/home/miniconda3/lib/python3.9/json/encoder.py", line 438, in _iterencode
    o = _default(o)
  File "/root/.local/lib/python3.9/site-packages/spikeinterface/core/core_tools.py", line 96, in default
    if np.issctype(obj):  # Cast numpy datatypes to their names
  File "/root/.local/lib/python3.9/site-packages/numpy/__init__.py", line 397, in __getattr__
    raise AttributeError(
AttributeError: `np.issctype` was removed in the NumPy 2.0 release. Use `issubclass(rep, np.generic)` instead.

```

alejoe91, 2024-06-21:
Yes for that you'll need to wait until the next release ;) #3061

Hopefully out on Monday

samuelgarcia, 2024-06-21:
Maybe you can try to do
```python
run_sorter(..., extra_requirements="numpy==1.26", docker_image=True, ...)
```
not sure at all if this works but this could solve temporary the issue because we set the boundaris numpy<2 in pypi release

alejoe91, 2024-06-21:
It might need to be a list: `["numpy==1.26.4"]`

taningh86, 2024-06-21:
@samuelgarcia i checked with pip show numpy and it is version 1.26.4. But I still tried and it forces pypi install of numpy 1.26.4 and it works. So the temporary solution is indeed working. @alejoe91 would you be updating the new release here also in addition to #3061 ? Just curious.

alejoe91, 2024-06-21:
The new release to come (0.101.0) is already fixed;)

h-mayorquin, 2024-06-21:
That's great @samuelgarcia !

taningh86, 2024-06-21:
@alejoe91 That's pretty quick and thanks!! You recommend going back to 101.0? Thats what I was using before. Besides some bug fixes, its out of my scope and depth to know what all improvements 101.0 has over 100.7. Is it worth currently to use 101.0 over 100.7? Sometimes going back and forth in versions gets a bit tiring ;)

zm711, 2024-06-21:
I would just recommend staying on 0.101.0/ main. I `SortingAnalyzer` is a huge benefit and so getting things working on main will probably have both a workflow + cognitive benefit in the long run for using this package.

taningh86, 2024-06-21:
@zm711 So I will need to uninstall 100.7 and reinstall 101.0 spikeinterface directly from the source or I can just install it without uninstalling 100.7?

taningh86, 2024-06-21:
Sorry to say this but this experience of going back and forth between versions has been a little frustrating. I was using 101 version which was great and used it for some months but then gave the numpy error, so after the discussion here I switched to 100.7 which did not get rid of the error but trying run_sorter (...extra_requirements=["numpy==1.26"]) recommendation from @samuelgarcia worked. Then I uninstalled 100.7 after hearing from @alejoe91 that 0.101.0 is fixed. Now I have been trying to install version 101 and have tried both `pip install spikeinterface[full,widgets]` and directly from the source ```
git clone https://github.com/SpikeInterface/spikeinterface.git
cd spikeinterface
pip install -e .
cd ..
```     However, both results in version 100.7. Additionally, now it looks like the package for matplotlib_base is corrupted and I am unable to plot anything. This is another fixing task I am working on now. The impetus for researchers to also become coders and troubleshooters is pretty strong these days. Though I think its good in the long run but I also strongly wish there is a relatively straightforward system or a plug and play system which is ready for analysis and don't need much fixing or troubleshooting. That seems like a more efficient way for us bench researchers. You guys are doing amazing work and these fixes are more attuned to your skillsets but for us it is more challenging to stay motivated to troublshoot.

taningh86, 2024-06-21:
Some texts did not make it in the last message. Anyways, it is still version 100.7 for both flavors of install and I have a matplotlib issue now. How do I get version 0.101?

zm711, 2024-06-21:
Sorry this is frustrating. NumPy 2.0 has affected quite a few packages and now we all have to work on trying to fix stuff. People actively updating/changing things in the past week are caught in the struggle for the package.

For your specific issue we aren't sure how it happened. It should have only tried to update numpy if you or someone else on the workstation did a 
```
pip install -U
```
or similar with Conda. The issue is when numpy was updated it broke everything for you.  Is this a shared workstation that someone else may have accidentally used? Or did you recently update anything?

 What I would do is

```
git clone https://github.com/SpikeInterface/spikeinterface.git
cd spikeinterface
pip install -e .[full,widgets]
cd ..
```

Notice you're missing the full,widgets with the . which will do the full install. Could you try that? And see if that gets us back up to speed.

taningh86, 2024-06-21:
@zm711 I am sure this is quite a work on you guys end. I am sorry that this has also effected everyone's work. 
I did not do any updates using `pip install -u`. However, since it gave me the numpy error I did uninstall and reinstall many times following the discussion here and to make things work. Not sure if that worsened my issue. 
Though this computer is shared the other person uses R and I don't think he updated anything. However, he uses DeeplabCut and may have done some updates on oen of the common packages. However, since he never uses the si_env package I don't think any of the updates he might have done effect me.
The installation code form the source you shared with [full,widgets] is not there in the spikeinterface page if you open the github page. it would be great if that can be updated there.
Also, after a reinstall do we need to also create a new environment with `conda env create --file full_spikeinterface_environment_windows.yml`? I have been doing that and not sure if that's necessary if I am not doing `conda remove --name si_env --all`.

taningh86, 2024-06-21:
I did as you suggested. But it still shows version 100.7.

zm711, 2024-06-22:
> I did as you suggested. But it still shows version 100.7.

@taningh86 I think first we have to make sure this is going into the correct environment. What does your terminal say? 

The version should be 0.101.0rc0 based on the pyproject.toml. Could you also open the pyproject.toml that you cloned with git clone and see what version it says inside the toml? 

> Also, after a reinstall do we need to also create a new environment with conda env create --file full_spikeinterface_environment_windows.yml? I have been doing that and not sure if that's necessary if I am not doing conda remove --name si_env --all.

This makes me think this could be contributing to the problem. Before doing anything with conda we need to make sure we are in the correct environment. So I wasn't clear before (that's my bad!)

```
conda activate si_env
git clone https://github.com/SpikeInterface/spikeinterface.git
cd spikeinterface
pip install -e .[full,widgets]
cd ..
```
You need to be in your environment. If both you and your research partner are fiddling with stuff in `(base)` as well as in your environment that could contribute to the issue.

If you are using the `full_spikeinterface_environment_windows.yml` it will attempt to install 0.100.7 because it does the most recent release. This is a limitation of being a pip package in something that is being so actively developed. We have this installation tips [rolling install](https://github.com/SpikeInterface/spikeinterface/blob/main/installation_tips/full_spikeinterface_environment_rolling_updates.yml)
to help users auto install main. So if you were trying to use the windows automatic environment install it could mess things up depending on the order you were doing things. 

Really everything will settle down in July-ish after the next full release.

h-mayorquin, 2024-06-24:
@alejoe91 now that there is a release, when should this be closed?

h-mayorquin, 2024-06-24:
OK, makes sense, yeah, try the new 0.100.8 release and I think this shoud fix it. Let us know.

taningh86, 2024-06-25:
Hi all! I followed the instructions of @zm711 and installed it from the source. The version I get is 0.101.0 not 0.100.8 as you guys were saying. I am assuming that's just a typo or it should be 0.100.8? Also I ran kilosort3 and it still gave the same attribute error -
`AttributeError: `np.issctype` was removed in the NumPy 2.0 release. Use `issubclass(rep, np.generic)` instead.`
I thought this was supposed to be taken care of. Below is the code I used without the `extra_requirements=["numpy==1.26"]`

```
sorting_KS25 = ss.run_sorter('kilosort3', recording_saved, 
                            output_folder=base_folder / 'results_KS3',
                            verbose=True, docker_image=True, 
                            **sorter_params)
```
Am I still suppose to use `extra_requirements= ["numpy==1.26']`? My understanding was that the recent fix made by @alejoe91 and others is to remedy this issue of numpy version. Am I wrong? or is it the new July update where this will be fixed? I am confused.

taningh86, 2024-06-25:
here's the step-by-step guide of what i did:
```
conda deactivate si_env
conda remove --name si_env --all
conda create -n si_env_100.8 python=3.9
conda activate si_env_100.8
git clone https://github.com/SpikeInterface/spikeinterface.git
cd spikeinterface
pip install -e .[full,widgets]
cd ..

pip install docker
```

alejoe91, 2024-06-25:
@taningh86 

Insetad of:
```
git clone https://github.com/SpikeInterface/spikeinterface.git
cd spikeinterface
pip install -e .[full,widgets]
cd ..
```
You should do:
```
pip install spikeinterface[full,widgets]==0.100.8
```

taningh86, 2024-06-25:
@alejoe91 ok, will do. To confirm, would that take care of the numpy 2.0 issue or I should still use` extra_requirements=["numpy==1.26.4"]`?

alejoe91, 2024-06-25:
That already takes care of the numpy issue

taningh86, 2024-06-25:
Ok. Thanks!

h-mayorquin, 2024-06-25:
As there is a solution to this and it will be fixed in the upcoming release for dev as well I am closing this.

closed 2024-06-25 by h-mayorquin