SpikeInterface/spikeinterface #3042: Segmentation fault (SpikeInterface, Docker, Kilosort4, Phy)

a-savoy, 2024-06-19:
I have no issues sorting and preparing the output to Phy using Kilosort4 (all default settings) via the Docker container with SpikeInterface. When I run phy trace-gui params.py there is no issue, but when I run phy template-gui params.py, I always see the following, regardless of the dataset:

16:33:44.806 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
Segmentation fault (core dumped)

If I run Kilosort4 outside of SpikeInterface, I do not have this issue.

h-mayorquin, 2024-06-19:
Which recording (format) are you using?

zm711, 2024-06-19:
@a-savoy could you run

```python
phy template-gui params.py --debug
```

That will give us the full stack trace for phy trying to load our files. Then we can see if we are missing a file in our version. Also KS4 is a bit ahead of our wrapper. So it is possible there is some mismatch that we still need to update.

JoeZiminski, 2024-06-19:
Additionally, it might be worth making a new environment from the `environment.yaml` file (if this was not how it was originally installed) (`conda env create --file=environment.yml`), sometimes this fixes the segfault on load.

zm711, 2024-06-19:
Only issue is I haven't updated the `environment.yml` for numpy yet.... I'm still working on getting the ci going on the phy repo. I keep getting RAM issues so I think my action might end up needing to be test by test. So If the yml doesn't work it is likely because I need to cap the numpy version.

a-savoy, 2024-06-20:
Thank you, I created an environment from environment.yml. The NumPy version was an issue, but worked after I changed the version to 1.24.1. However, I still get the segmentation fault. Here is the debug output:

> $ phy template-gui params.py --debug
> 14:14:35.547 [D] __init__:68          Start capturing exceptions.
> 14:14:35.731 [D] model:619            Loading spike clusters.
> 14:14:35.861 [D] model:569            No channel shank file found.
> 14:14:35.861 [D] model:692            Loading templates.
> 14:14:35.864 [D] model:720            Templates are sparse.
> 14:14:35.869 [W] model:667            Skipping spike waveforms that do not exist, they will be extracted on the fly from the raw data as needed.
> 14:14:35.870 [D] model:730            Loading the whitening matrix.
> 14:14:35.870 [D] model:434            Whitening matrix file not found.
> 14:14:35.870 [D] model:737            Loading the inverse of the whitening matrix.
> 14:14:35.872 [D] model:766            Loading features.
> 14:14:35.873 [D] model:781            Features are sparse.
> 14:14:35.873 [D] model:803            Loading template features.
> 14:14:35.874 [D] model:504            Load `cluster_group.tsv`.
> 14:14:35.875 [D] model:504            Load `cluster_si_unit_ids.tsv`.
> 14:14:35.876 [D] model:504            Load `cluster_channel_group.tsv`.
> 14:14:36.174 [D] context:100          Initialize joblib cache dir at `/phy/.phy`.
> 14:14:36.174 [D] context:101          Reducing the size of the cache if needed.
> 14:14:36.176 [D] base:102             Add filter `high_pass`.
> 14:14:36.177 [D] config:31            Load config file `/.phy/phy_config.py`.
> 14:14:36.177 [D] plugin:146           Loading 0 plugins.
> 14:14:36.180 [D] context:209          The file `/phy/.phy/new_cluster_id.pkl` doesn't exist.
> 14:14:36.225 [D] context:185          Save data to `/phy/.phy/spikes_per_cluster.pkl`.
> 14:14:36.309 [D] gui:463              Creating GUI.
> 14:14:36.320 [D] state:46             Load /.phy/TemplateGUI/state.json for GUIState.
> Segmentation fault (core dumped)

zm711, 2024-06-20:
This might be a Phy issue at this point. It seems like loading the gui state is failing. Do you want to open your issue over there and I will help you on the dedicated repo. This isn't a missing spikeinterface file, so I'll close this here, but happy to help over there.

closed 2024-06-20 by zm711

a-savoy, 2024-06-20:
Sounds good, I'll move it over there. Thanks for your help.