SpikeInterface/spikeinterface #3067: error message for kilosort 4 (installed) with default arguments

rat-h, 2024-06-24:
```python
Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import spikeinterface.full as si
>>> 
>>> si.__version__
'0.100.7'
>>> import kilosort as ks
>>> ks.__version__
'4.0.12'
>>> rc = si.load_extractor("continuous-ks4.0-test/preprocessed")
>>> sorting = si.run_sorter(sorter_name="kilosort4",recording=rc,output_folder="continuous-ks4.0-test/sorter-workdir")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "../.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py", line 175, in run_sorter
    return run_sorter_local(**common_kwargs)
  File "../.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/runsorter.py", line 225, in run_sorter_local
    SorterClass.run_from_folder(output_folder, raise_error, verbose)
  File "../.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/basesorter.py", line 293, in run_from_folder
    raise SpikeSortingError(
spikeinterface.sorters.utils.misc.SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "../.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/basesorter.py", line 258, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "../.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/external/kilosort4.py", line 197, in _run_from_folder
    ops = initialize_ops(settings, probe, recording.get_dtype(), do_CAR, invert_sign, device)
TypeError: initialize_ops() missing 1 required positional argument: 'save_preprocesed_copy'

Spike sorting failed. You can inspect the runtime trace in spikeinterface/continuous-ks4.0-test/sorter-workdir/spikeinterface_log.json.
>>> 
```

```json
{
    "sorter_name": "kilosort4",
    "sorter_version": "4.0.12",
    "datetime": "2024-06-24T11:07:19.173059",
    "runtime_trace": [],
    "error": true,
    "error_trace": "Traceback (most recent call last):\n  File \"../.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/basesorter.py\", line 258, in run_from_folder\n    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)\n  File \"../.local/apps/spikes/lib/python3.10/site-packages/spikeinterface/sorters/external/kilosort4.py\", line 197, in _run_from_folder\n    ops = initialize_ops(settings, probe, recording.get_dtype(), do_CAR, invert_sign, device)\nTypeError: initialize_ops() missing 1 required positional argument: 'save_preprocesed_copy'\n",
    "run_time": null
}
spikeinterface_log.json (END)
```

alejoe91, 2024-06-24:
We're making a release with the KS4 fixes today! #3061

alejoe91, 2024-06-24:
Released: https://pypi.org/project/spikeinterface/0.100.8/

closed 2024-06-24 by alejoe91