SpikeInterface/spikeinterface #3069: Spike sorting error

AbhiSwamiUConn, 2024-06-24:
When I try running a sorter with either openephys or plexon files, I get an error stating that I need at least one array to concatenate.  Previously, this error did not show up and the files I tried ran without issue.  Is this something on my end?
```
SpikeSortingError                         Traceback (most recent call last)
Cell In[3], line 28
     26 print(filtered_recording.is_filtered())
     27 #sort and analyze
---> 28 sorting = ss.run_sorter(sorter_name=sorter_name, recording=recording, output_folder=output_folder1, remove_existing_folder=True)
     29 Sanalyzer = si.waveform_extractor.extract_waveforms(recording=recording, sorting=sorting, folder = output_folder2, overwrite= True)
     30 recording.analyze_extractor(Sanalyzer)

File ~\anaconda3\Lib\site-packages\spikeinterface\sorters\runsorter.py:175, in run_sorter(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, delete_container_files, with_output, **sorter_params)
    168             container_image = singularity_image
    169     return run_sorter_container(
    170         container_image=container_image,
    171         mode=mode,
    172         **common_kwargs,
    173     )
--> 175 return run_sorter_local(**common_kwargs)

File ~\anaconda3\Lib\site-packages\spikeinterface\sorters\runsorter.py:225, in run_sorter_local(sorter_name, recording, output_folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, **sorter_params)
    223 SorterClass.set_params_to_folder(recording, output_folder, sorter_params, verbose)
    224 SorterClass.setup_recording(recording, output_folder, verbose=verbose)
--> 225 SorterClass.run_from_folder(output_folder, raise_error, verbose)
    226 if with_output:
    227     sorting = SorterClass.get_result_from_folder(output_folder, register_recording=True, sorting_info=True)

File ~\anaconda3\Lib\site-packages\spikeinterface\sorters\basesorter.py:293, in BaseSorter.run_from_folder(cls, output_folder, raise_error, verbose)
    290         print(f"{sorter_name} run time {run_time:0.2f}s")
    292 if has_error and raise_error:
--> 293     raise SpikeSortingError(
    294         f"Spike sorting error trace:\n{log['error_trace']}\n"
    295         f"Spike sorting failed. You can inspect the runtime trace in {output_folder}/spikeinterface_log.json."
    296     )
    298 return run_time

SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "C:\Users\User\anaconda3\Lib\site-packages\spikeinterface\sorters\basesorter.py", line 258, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "C:\Users\User\anaconda3\Lib\site-packages\spikeinterface\sorters\internal\spyking_circus2.py", line 118, in _run_from_folder
    recording_f = zscore(recording_f, dtype="float32")
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\anaconda3\Lib\site-packages\spikeinterface\preprocessing\normalize_scale.py", line 296, in __init__
    random_data = get_random_data_chunks(recording, return_scaled=False, **random_chunk_kwargs)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\User\anaconda3\Lib\site-packages\spikeinterface\core\recording_tools.py", line 569, in get_random_data_chunks
    return np.concatenate(chunk_list, axis=0)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: need at least one array to concatenate

Spike sorting failed. You can inspect the runtime trace in C:\Users\User\Desktop\Abhinav\files1/spikeinterface_log.json.

```
[image]

h-mayorquin, 2024-06-24:
Maybe another check to add in general is that the memmaps have data (with shape I guess). From here we don't know if some preprocessing step fail or if the data just failed to reload.

AbhiSwamiUConn, 2024-06-24:
I get this when I try to print the recording:` BandpassFilterRecording: 16 channels - 30.0kHz - 0 segments - 0 samples - 0.00s (0.00 ms) int16 dtype - 0.00 B`.  My filter was set between 250 and 2500 hertz.

alejoe91, 2024-06-24:
That's the problem! You have no samples in your recording! Can you print the recording before the bandpass filter? You did you load it?

AbhiSwamiUConn, 2024-06-24:
I definitely loaded the recording.  When loading before the filter, I get: ```
SelectSegmentRecording: 16 channels - 30.0kHz - 0 segments - 0 samples - 0.00s (0.00 ms) int16 dtype - 0.00 B```.  Is the issue that I am trying to access segments in a recording that does not have multiple segments?

AbhiSwamiUConn, 2024-06-24:
I was utilizing user input and I had an statement set up where if there are no recording segments, the step of selecting specific segments would be skipped.  This statement appears to be getting skipped.  When I try running my script with the change it works.  Sorry for taking your time.

h-mayorquin, 2024-06-24:
You don't even have data in that recording, check the number of samples.

AbhiSwamiUConn, 2024-06-24:
I do have data.  I unknowingly passed an empty list to the segment recording extractor.  Here is the recording with all data: ```
OpenEphysBinaryRecordingExtractor: 16 channels - 30.0kHz - 1 segments - 18,272,400 samples  609.08s (10.15 minutes) - int16 dtype - 557.63 MiB
```

h-mayorquin, 2024-06-24:
I meant at the point of the print above. It says you have zero samples.

AbhiSwamiUConn, 2024-06-24:
Yes, and I have stated it was fixed.  Here is the recording after the check for multiple segements and after it has been filtered: `BandpassFilterRecording: 16 channels - 30.0kHz - 1 segments - 18,272,400 samples 609.08s (10.15 minutes) - int16 dtype - 557.63 MiB`

closed 2024-06-24 by AbhiSwamiUConn

h-mayorquin, 2024-06-24:
Glad it worked for you : )