SpikeInterface/spikeinterface #3078: Random Spikes and Waveforms

atombysx, 2024-06-25:
Hi,

I'm transitioning from waveform extractor to SortingAnalyzer.

But I'm having problem accessing the saved files.

In my pipeline, I integrate spikeinterface output with UnitMatch that I load the waveforms for each unit (corresponding segment index and waveform) at a time with information from sparsity.npy.

Now I see that the waveforms are computed corresponding to random_spikes_indices.npy which corresponds to the sample index and the waveforms are saved altogether in one big .npy file rather than individual unit.

But I'm having difficulties to find what unit each waveform is corresponding to... and I couldn't find anything documentation about this.

chrishalcrow, 2024-06-25:
Hi @atombysx 

I think this was changed so that we could more easily load the large waveforms file lazily, which is much more memory efficient for the calculations that involve waveforms. Hope it's not too much of a nuisance. 

The .npy file is just all the concatenated waveforms, so you're right: it's not simple to split it into individual units. You can get the waveforms of an individual unit using the `get_waveforms_one_unit` method on a sorting_analyzer's waveform extension. That works as follows:
```
unit_5_waveforms = sorting_analyzer.get_extension("waveforms").get_waveforms_one_unit(unit_id=5)
```
etc. Then you could export these separately for UnitMatch. Will that work for your problem?

Also, if we ever need _all_ the waveforms, you can force random_spikes to select to _all_ spikes using `si.compute_random_spikes(sorting_analyzer, method = "all")`

atombysx, 2024-06-25:
> Hi @atombysx
>
> I think this was changed so that we could more easily load the large waveforms file lazily, which is much more memory efficient for the calculations that involve waveforms. Hope it's [...]

Hi Chris, thanks!

I guess I can also refer to ['unit_index'] in spikes.npy to find the corresponding units.

closed 2024-06-25 by atombysx

chrishalcrow, 2024-06-25:
Yeah, there's a few options. There's also `sorting_analyzer.get_extension("random_spikes").get_random_spikes()` which gives you the spiketrain of the randomly chosen spikes. You can see the mess of things that `get_waveforms_one_unit` uses here: https://github.com/SpikeInterface/spikeinterface/blob/d963aa26e31a9b6bc0bea2a4973f76d4dbafb062/src/spikeinterface/core/analyzer_extension_core.py#L227

Thanks for bringing it up - I'll add some documentation about this at some point :)