SpikeInterface/spikeinterface #3083: `correct_motion_on_peaks` is not included in the API docs

JoeZiminski, 2024-06-25:
Not sure why, I will look into this ASAP.

JoeZiminski, 2024-06-25:
Same with `load_motion_info`, maybe a few from the motion correction are not there.

chrishalcrow, 2024-06-26:
Just having a quick look at this: am I right in thinking that we maintain the `api.rst` file? I always thought sphinx did this for us!

samuelgarcia, 2024-06-26:
Sphinx could do it but then we need a an automodule/autofunction/autoclass everywhere (mainly adding doc in all init.py) with the same probality to forget to add it.
I think centralized is not so bad but we very often forgot to add new function and classes

chrishalcrow, 2024-07-19:
Hello, [Motion Correction](https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.sortingcomponents.motion.correct_motion_on_peaks) is now there. But `load_motion_info` and `save_motion_info`. Should these be in the `preprocessing` api? If yes, I'll add them there.

PR #4057 (2025-07-14): Add motion functions to api docs
Fixes #3083.

The second half of missing api functions. Now added to the api.rst.

Also a tiny fix. The semantic (that Chris and I discussed) for saying backward compatible would be to save that it is for versions greater than what we are being compatible to. So I think we want to actually list this compatibility as > 100 rather than < 101. Otherwise I think we would rewrite that whole statement.
closed 2025-07-14 by alejoe91