SpikeInterface/spikeinterface #3086: Standardise order of recording, sorting args

jakeswann1, 2024-06-26:
Hi all! I've noticed that for different functions which require both a recording and a sorting, the order in which these should be passed is not consistent. E.g. `si.create_sorting_analyzer(sorting, recording)` but `si.estimate_sparsity(recording, sorting)`

Not exactly high priority, but it might be nice to standardise this order in the API at some point? Although I can also see how it might be better to have the first argument be the most relevant object for whatever that function is doing

h-mayorquin, 2024-06-26:
I think you should use keyword arguments for this:

```
si.create_sortying_analyzer(sorting=sorting, recording=recording).
```

I think relying on position is a bad practice, if we remove  a parameter at some point the order will shift and you will get an error. With the keyword, unless we remove that one you won't get it. Unless we remove that one and the error will be clear.

zm711, 2024-06-26:
I take Jake's point though. Although we all preach kw I often find myself doing positional when working in the terminal out of laziness. Having a system of ordering those for positional is just more consistent and lowers mental burden. That way we aren't telling the end-user it's your fault for using positional arguments.

h-mayorquin, 2024-06-26:
Fair enough.

It did not occur to me that people are working from the terminal. I am in seldom in environments in situations where typing is a bottleneck so there is my bias.

This is A BIG BREAKING CHANGE though.

zm711, 2024-06-26:
Yeah it is true this would probably be too big a change. But if I'm not mistaken the `WaveformExtractor` and the `SortingAnalyzer` flipped the order so people trying to switch over with 101 release (if they use positional) are going to be annoyed.

h-mayorquin, 2024-06-26:
Maybe that one we can get done before the release, @samuelgarcia, what do you think? Seems like a good quality of life improvement to make the order consistent between waveform extractor and sorting.

samuelgarcia, 2024-06-26:
Hi.
The flip was really on purpose.
In early dev, we very often put the recording as first but this was mistake, in my hubble point of view. This reason is that many function are sorting centric : do something with the sorting. So logically it has to be the first args no ?
SortingAnalyzer is about sorting, so having sorting fist was really important for me.
WaveformExtactor was the wrong I think.
Also I think I am agree with Jake : estimate_sparsity(recording, sorting) should be estimate_sparsity(sorting, recording) because it is also sorting centric.
Having something more logical towards version 1 (may 2027) is more important that having the same signature as before. no ?
What could be done is : to add some magic to flip args when they are accidentaly reverted but not sure this is a good thing.

zm711, 2024-06-26:
I'm honestly fine with either order, but consistency is most important to me. So I think if we agree that all arguments will be sorting first (when applicable). But we should do that for sparsity now, no? 101 already has a breaking change so get this one in their too?

zm711, 2024-06-27:
Thanks Chris! `has_exceeding_spikes` and `SpikeRetriever` are super internal so they would actually be easy to fix without risk of breaking others code. So that means only sparsity would be breaking I guess.

h-mayorquin, 2024-06-27:
Thanks Christ for actually counting : )

zm711, 2024-06-27:
> Thanks Christ for actually counting : )

You're going to give Chris a big head by making that equivalence :P

PR #3094 (2024-06-28): Reordering recording, sorting args
Would fix #3086, if we want to break `estimate_sparsity`
closed 2024-06-28 by samuelgarcia