SpikeInterface/spikeinterface #3087: sorting_analyzer not incorporated in version 0.100.8

taningh86, 2024-06-26:
Hi,
This issue could have been tagged in my last issue post about Numpy error with ver 0.101.0 but it is closed now so making a new issue here. In version 0.100.8 sorting_analyzer is not incorporated yet. So I have been using the waveform extractors and postprocessing modules instead. Not sure if this was how you guys also expected, but just letting you guys know. 

Thanks!

alejoe91, 2024-06-26:
yes that's expected! We're ironing out the last details and we'll make a release with the analyzer by July 13th :)

taningh86, 2024-06-26:
Ok. Sounds good. Thanks!

zm711, 2024-07-11:
Cool I'll close this now since the sorting_analyzer will never be incorporated into 0.100.8.

closed 2024-07-11 by zm711