SpikeInterface/spikeinterface #3093: Add ticks to all arguments referenced in docstrings

zm711, 2024-06-27:
@JoeZiminski @chrishalcrow @h-mayorquin 

Discussion in #3068 (specifically this [thread](https://github.com/SpikeInterface/spikeinterface/pull/3068#discussion_r1657040581))

In the library we've been a bit inconsistent about referencing other arguments inside docstrings. We had originally agreed on ticks, so for example

```python

def a_function(a, b, c):
    """
    Parameters
    ----------
    a : array
        the data
    b : bool
        whether to scale the data in `a`
    """
```
Does that sound good to everyone and something to add while working on numpydoc conventions?

alejoe91, 2024-06-28:
I like your definition of fun @chrishalcrow eheheheh

h-mayorquin, 2024-07-03:
So, we should use the one that is used on markdown, that's `here` right?

zm711, 2024-07-10:
Yes @h-mayorquin the standard ` that is also used in markdown.