SpikeInterface/spikeinterface #3099: Regarding the issues of "si.read_openephys"

TsungChihTsai, 2024-06-27:
Before I perform this way to analysis, it works. In the beginning, I found another issue about asking me to add "load_sync_channel= False". I tried it, but it remain the same issue. Therefore, I update my pip install spikeinterface to the lastest version. I analyzed the open ephys binary data, I encounter this issue. I also attempted to uninstall spikeinterface, and then install my original version I have ever used "pip install spikeinterface[full,widgets]==0.99.0" and "spikeinterface-gui>=0.4.1". And, it still show the same issue as below: 

recording = si.read_openephys("D:/20240403_ASI/WT_iso_2024-04-03_11-12-01_001", stream_name="Record Node 105#File_Reader-100.Data",experiment_names ="experiment 1")

[image]
[image]

Could you please guide me to modify my path?

alejoe91, 2024-06-27:
Hi,

What is the content of the folder? Can you print it also with subfolders? Did you delete any folder manually?

TsungChihTsai, 2024-06-27:
Yes, I can show you. 
**continuous subfolders**
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\continuous\File_Reader-100.Data\continuous.dat 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\continuous\File_Reader-100.Data\input_flag.txt 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\continuous\File_Reader-100.Data\sample_numbers.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\continuous\File_Reader-100.Data\timestamps.npy 

**recording1 subfolders**
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\structure.oebin 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\sync_messages.txt 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\continuous\input_flag.txt 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\continuous\File_Reader-100.Data\continuous.dat 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\continuous\File_Reader-100.Data\input_flag.txt 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\continuous\File_Reader-100.Data\sample_numbers.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\continuous\File_Reader-100.Data\timestamps.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\events\File_Reader-100.Data\TTL\full_words.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\events\File_Reader-100.Data\TTL\sample_numbers.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\events\File_Reader-100.Data\TTL\states.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\events\File_Reader-100.Data\TTL\timestamps.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\events\MessageCenter\sample_numbers.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\events\MessageCenter\text.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\events\MessageCenter\timestamps.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 1\clusters.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 1\electrode_indices.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 1\sample_numbers.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 1\timestamps.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 1\waveforms.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 10\clusters.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 10\electrode_indices.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 10\sample_numbers.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 10\timestamps.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 10\waveforms.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 11\clusters.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 11\electrode_indices.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 11\sample_numbers.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 11\timestamps.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 11\waveforms.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 12\clusters.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 12\electrode_indices.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 12\sample_numbers.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 12\timestamps.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 12\waveforms.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 13\clusters.npy 
[... cut ...]
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 26\waveforms.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 27\clusters.npy 
D:\20240403_ASI\WT_iso_2024-04-03_11-12-01_001\Record Node 105\experiment1\recording1\spikes\Spike_Detector-104.Data\Electrode 27\electrode_indices.npy 

No, I don't delete anything. It's the same folder I've ran spikeinterface.

alejoe91, 2024-06-27:
Maybe `experiment 1` should be `experiment1` (no space)?

TsungChihTsai, 2024-06-27:
I revised "recording = si.read_openephys("D:/20240403_ASI/WT_iso_2024-04-03_11-12-01_001", stream_name="Record Node 105#File_Reader-100.Data",experiment_names ="experiment1")", it still shows the same error. 

[image]
[image]

alejoe91, 2024-06-27:
Now the error is different. Can you add the `load_sync_channel=False` argument?

TsungChihTsai, 2024-06-27:
I use another laptop at home now. I don't do any update for this laptop (I think it is not caused by update the module, but it happened today). This is different data. Because previous one is huge, I just keep a small one in my laptop.  **I encounter same issue, too.** So, I added the **load_sync_channel=False** argument, but still shows an error shows. op. 

---- laptop at home
I ran a data is already sorted by kilosort3. 

recording = si.read_openephys("D:\WT_ASI_2024-04-25_10-18-03_001", stream_name="Record Node 113#OE_FPGA_Acquisition_Board-119.Rhythm Data",experiment_names ="experiment1", **load_sync_channel=False**)

**And, it works.** 

If I chose a new data I've never used before. 
[image]
[image]
[image]
[image]
It shows another issue which is due to "recording_saved = recording_sub.save(folder=output_folder / "preprocessed", **job_kwargs) "

====I also use "pip list" to check the version regarding to spikeinterface: 
spikeinterface                0.99.0
spikeinterface-gui            0.7.0
probeinterface                0.2.19
numpy                         1.23.5 ( list this one, because I found I cannot ran this because the numpy version is higher than 2.0.0 in the beginning)
====For tracing the version I've used well before. 

----I will check computer in the lab
Tomorrow, I will check the desktop in the lab. Maybe, the new one issue is related to the data (**If I find this is true, I will raise another issue for that**) , not the issue about this time. 

And, I will let you know what happened. Thank a million.

alejoe91, 2024-06-28:
Can you please upgrade to the latest spike interface? We fixed a bunch of issues in the recent releases!

chrishalcrow, 2024-06-28:
If you keep getting errors about `BrokenProcessPool`s and things like that, then the problem is related to multiprocessing. A good idea is to start with `n_jobs=1`, then move on to multiprocessing once everything is working with one job. But as @alejoe91 says, the multiprocessing has a better chance of working easily with the newest version of spikeinterface!

TsungChihTsai, 2024-06-28:
Thanks for your suggestion.

I attempted to update spikeinterface in the lab desktop
[image]
Successfully installed spikeinterface-0.101.0rc0

And, add load_sync_channel=False

But, it still have an issue,
[image]

I think maybe I update in a wrong way?

alejoe91, 2024-06-28:
@TsungChihTsai can you send us the data? A google drive link will work!

TsungChihTsai, 2024-06-28:
I save the original data without modified anything in google drive. 

The link is as below:
https://drive.google.com/drive/folders/1RFGTl8gtUbGqRShDMLPfM5rA5AOrzU6L?usp=sharing

If you find any problem, please let me know.

Thanks for your help.

----

I also create a new spikeinterface environment (https://github.com/SpikeInterface/spikeinterface/tree/main/installation_tips) as following steps: 
cd C:\code\spikeinterface_installation
conda env create --file full_spikeinterface_environment_windows.yml
conda activate spikeinterface_env

git clone https://github.com/SpikeInterface/spikeinterface.git
cd spikeinterface
pip install -e .
cd ..

python cleanup_for_windows.py
pip install notebook 
jupyter notebook 

But, it still show the same error. 

----

TsungChihTsai, 2024-06-28:
@alejoe91 
This data (https://drive.google.com/drive/folders/1RFGTl8gtUbGqRShDMLPfM5rA5AOrzU6L?usp=sharing) can run "recording = si.read_openephys("D:\WT_ASI_2024-04-25_10-18-03_001", stream_name="Record Node 113#OE_FPGA_Acquisition_Board-119.Rhythm Data",experiment_names ="experiment1", load_sync_channel=False)" on my laptop at home.

@chrishalcrow Yes, I change n_jobs=10 to n_jobs=1. And, it can continue the  "recording_saved = recording_sub.save(folder=output_folder / "preprocessed", **job_kwargs) "!
[image]

When I ran the step of sorting, it show the original issue I met.

E=kilosort3
    sorting = si.run_sorter(E, recording_saved, 
                                output_folder=output_folder/ E / 'results',
                                docker_image=True,verbose=True, **sorter_params)  

[image]

It show the original issue I met.
AttributeError: `np.issctype` was removed in the NumPy 2.0 release. Use `issubclass(rep, np.generic)` instead. 

When I face this situation in that time, I updated my numpy on the desktop. Then, it shows "keyerror=0"

---- (latest version of spikeinterface)
[image]

I also attempt to create a new environment and update the latest version of spikeinterface on my laptop, even though I added "load_sync_channel=False". Still have the issue for "ValueError: SYNC channel is not present in the recording. Set load_sync_channel to False"

[image]

TsungChihTsai, 2024-06-30:
"ValueError: SYNC channel is not present in the recording. Set load_sync_channel to False"
The issue is from "..\miniconda3\envs\si_env\Lib\site-packages\neo\rawio\openephysbinaryrawio.py"
I revised, and force it as "load_sync_channel=False"
[image]
However, I am not sure if it will have an adverse effect? (I guess if someone have synchronized data, it still have an issue) 
Finally, I can run extractor. 

When I run kilosort3 in docker, it shows another issue: 
"AttributeError: np.issctype was removed in the NumPy 2.0 release. Use issubclass(rep, np.generic) instead.", it's related to "spikeinterface/kilosort3-compiled-base" in docker. Therefore, I update **spikeinterface==0.100.8**, docker.exe to v4.31.1 and change "job_kwargs = dict(n_jobs=20, chunk_duration="1s", progress_bar=True)"

[image]

When I run kilosort4 in docker: 
I got "AttributeError: Bad parameters: ['n_jobs', 'chunk_duration', 'progress_bar']", I **change parameters to "si.get_default_sorter_params("kilosort4")"**
[image]

I resolve these issues.

alejoe91, 2024-06-30:
Let's still leave this open, since the open ephys error needs to be checked. I'll try to take a look at the data you shared on Monday.

TsungChihTsai, 2024-06-30:
Thanks a million.

alejoe91, 2024-07-01:
@TsungChihTsai I requested access to the drive folder

alejoe91, 2024-07-02:
@TsungChihTsai it works on my side:

This is with the current main version:
[image]

And this is with the latest stable release (0.100.8):
[image]

@zm711 maybe this could be a Windows problem?

zm711, 2024-07-02:
Happy to test on Windows. But @TsungChihTsai you mentioned your neo version is 0.12.0. Could you update to 0.13.1? Or install from source. We've worked on this IO a bunch since the last release. If it is still not working with neo updated then I'll see if it is a Windows problem :)

TsungChihTsai, 2024-07-02:
pip install neo==0.13.1

[image]

It still has this issue. 
[image]

alejoe91, 2024-07-02:
Can you just try the code from my screenshots here? https://github.com/SpikeInterface/spikeinterface/issues/3099#issuecomment-2202087760

This is using the folder you shared

TsungChihTsai, 2024-07-02:
Yes, it works! Thanks!

zm711, 2024-07-02:
Sorry, I missed this. I just requested access, but if it works then I'm not needed :)

alejoe91, 2024-07-02:
Ok! closing then!

closed 2024-07-02 by alejoe91