SpikeInterface/spikeinterface #3106: How to plot correlograms after removing the timetamps of inter-spike frequency >100 Hz?

TsungChihTsai, 2024-06-30:
Because we got the sorting data are higher frequency, we'd like to exclude the abnormal timestamps. After curated waveforms, we export the timestamps and remove the timetamps of inter-spike frequency >100 Hz and export the filtered timestamps. And, we save the filtered timestamp as csv file. Each column is each units. The values are timetamps. 

[image]
Could I use "ccgs, bins = si.compute_correlograms", and "si.plot_crosscorrelograms" to process and plot correlograms?

I saw https://github.com/SpikeInterface/spikeinterface/issues/2668, but we need to obtain sorting data after I filtered timestamp. 

One possibility is that we use some extractor to put the data (we can obtain the data of filtered timestamps and filtered waveform) as a sorting result. 
 
Or, maybe spikeinterface have internal module to remove inter-spike frequency >100 Hz by each timestamps, not use curated waveforms to remove the units.

JoeZiminski, 2024-07-01:
Hi @TsungChihTsai, I think one approach would be as you mention to load your .csv file into a `Sorting` object. @zm711 wrote a guide on this [here](https://spikeinterface.readthedocs.io/en/latest/how_to/load_your_data_into_sorting.html). I think the easiest approach would be to:

1) load data into python with `pd.read_csv()`
2) convert the time to sample index based on your sampling frequency
3) Use the unit labels + spike times (in samples) to make a `Sorting` object and use this to make correlograms.

TsungChihTsai, 2024-07-01:
Thanks for your guidance. 

I attempted to use "With a unit dictionary"
[image]
After loading data into python, I converted the time to sample index based on your sampling frequency 

[image]

[image]
...
[image]

The number of unit_ids (Unit0 to Unit61) and the sample index are right. 

However, when I plot the plot_crosscorrelograms, I found all the ccgs are 0 (It is abnormal), bins are normal
[image]
[image]

I think I do something wrong. Could you please told me how could I do for the next step?

zm711, 2024-07-02:
I'm not sure here. But one mistake you've made is you've overwritten NumpySorting. so instead of 

```python
NumpySorting = NumpySorting.from_unit_dict(...)
```

you should do something like

```python
filtered_sorting = NumpySorting.from_unit_dict()
```

could you try that and see what happens.

I'm also trying to understand what you mean inter-spike frequency > 100 Hz. Could you describe that a bit more. I'm trying to understand what you are actually doing to filter your data.

alejoe91, 2024-07-02:
@TsungChihTsai do you mean just firing rate?

TsungChihTsai, 2024-07-02:
**Yes, it works! 
filtered_sorting = NumpySorting.from_unit_dict()
I just use original data as below:** 
[image]

**Sorry to confuse you.** 
**Like this, just change Inter-spike-interval (ISI) to Inter-spike-frequency
100 Inter-spike-frequency => 10 ISI**

[image]
**Like this, just change Inter-spike-interval (ISI) to Inter-spike-frequency
100 Inter-spike-frequency => 10 ISI**

**Can I use this to do so?** 
[image]
[image]

TsungChihTsai, 2024-07-02:
Original one from our data
[image]

Another one, I extract from original one
[image]
[image]

I think it should be the same pattern, but not.

JoeZiminski, 2024-07-02:
Thanks @TsungChihTsai, glad it is loading okay. Could you please post the full code used to generate both correlograms that should match, and copy the text into github (you can use three backticks ``` ``` like this and put the code int the middle).

alejoe91, 2024-07-02:
@TsungChihTsai 

What you're trying to do (filtering a spike sorting output based on some quality metrics) is well documented in the get started tutorial:
https://spikeinterface.readthedocs.io/en/latest/get_started/quickstart.html

and in the quality metrics curation tutorial:
https://spikeinterface.readthedocs.io/en/latest/tutorials/qualitymetrics/plot_4_curation.html

TsungChihTsai, 2024-07-02:
@JoeZiminski 
Original one
[image]
[image]

using NumpySorting.from_unit_dict
[image]
[image]
[image]

I found the ccgs are different. Therefore, I attempted to check the source.
I extract the timestamps from spikeinterface
[image]

Save in csv 
[image]
I found they are the same value and order. 
----
@alejoe91 I'd like to remove the ISI<10 for each timepoint, but not using "quality metrics curation" for each unit. I mean I don't want to remove some units due to "isi_violation", because the other feature are good in some unit. Therefore, I tried to remove some timepoints due to the ISI is too short (Inter-spike-frequency is more than 100 Hz) in my desired unit.

alejoe91, 2024-07-02:
@TsungChihTsai got it!

Then we do have a function for that:
```
import spikeinterface.curation as scur

sorting_remove_isi = scur.remove_duplicated_spikes(sorting, censored_period_ms=10)
```
This will remove all spikes within 10ms.

TsungChihTsai, 2024-07-02:
Thanks a lot. Sorry to confuse you. Inter-spike-frequency, this express is wrong, I should use instant frequency to express the frequency between two spikes.

JoeZiminski, 2024-07-02:
@TsungChihTsai many thanks, maybe I am misunderstanding, is the difference not because in the first example all spikes are included, in the spikes with small ISI have been filtered out? 

Unfortunately it's a little hard to tell why these are different because it's not clear where the first example came from, how the preprocessing and sorting was run. Are both examples definitely using exactly the same data? If you could upload your `.csv` file and copy and paste the code that you are using to load into a `NumpySorting` and make the correlogram. I'd be happy to check this is working as expected.

TsungChihTsai, 2024-07-02:
Yes, it is the same. I can show this to let you check my original data for correlogram : 
[correlograms.zip](https://github.com/user-attachments/files/16070425/correlograms.zip)

My pleasure. 
[spike_times_4_each_unit(column).csv](https://github.com/user-attachments/files/16070370/spike_times_4_each_unit.column.csv)

The code is: 
import numpy as np
from pathlib import Path
from spikeinterface import extractors as se
from spikeinterface.core import NumpySorting

# Load spike times from CSV
spike_times_path = Path(output_folder/E/"spike_times_4_each_unit(column).csv")
spike_times_data = np.genfromtxt(spike_times_path, delimiter=',')
spike_times_data = np.transpose(spike_times_data) 
# Filter spike times to remove trailing zeros
last_non_zero_idx = np.max(np.nonzero(spike_times_data)) + 1
filtered_spike_times = spike_times_data[:, :last_non_zero_idx]* 30000/1000
# Create unit IDs
unit_ids = list(range(filtered_spike_times.shape[0]))
# Create times list
times_list = [filtered_spike_times[i][filtered_spike_times[i] > 0] for i in range(filtered_spike_times.shape[0])]

# Create dictionary for from_unit_dict
units_dict = {str(unit_id): np.array(times_list[i]) for i, unit_id in enumerate(unit_ids)}
print(units_dict)
# Create NumpySorting extractor using from_unit_dict
filtered_sorting =  NumpySorting.from_unit_dict(
    units_dict_list=units_dict,
    sampling_frequency=30_000.0
)
print('ccgs :',ccgs)
print('bins :', bins)
si.plot_crosscorrelograms(filtered_sorting, unit_ids=filtered_sorting.unit_ids[::10], figsize=(10, 6))
[correlograms.zip](https://github.com/user-attachments/files/16070407/correlograms.zip)

alejoe91, 2024-07-02:
> @TsungChihTsai got it!
> 
> Then we do have a function for that:
> 
> ```
> import spikeinterface.curation as scur
> 
> sorting_remove_isi = scur.remove_duplicated_spikes(sorting, censored_period_ms=10)
> ```
> 
> This will remove all spikes within 10ms.

@TsungChihTsai have you tried this?  https://github.com/SpikeInterface/spikeinterface/issues/3106#issuecomment-2203423042

TsungChihTsai, 2024-07-02:
[image]
It works! Thanks a million.

alejoe91, 2024-07-02:
Closing then :)

closed 2024-07-02 by alejoe91