SpikeInterface/spikeinterface #3110: This meta file version of spikeglxis not implemented

twheatcroft07, 2024-07-01:
Hi there,

I have an issue when I try to run the code below. I get the error 'NotImplementedError: This meta file version of spikeglxis not implemented'. I only see this error when I have the lf.bin and lf.meta files in that folder, and not when I only have the other files (screenshot of list of files attached). 
[image]

`spikeglx_folder '20240606_M230_g0\catgt_20240606_M230_g0'`
`stream_names, stream_ids = si.get_neo_streams('spikeglx', spikeglx_folder)`

Can you offer any advice?

Thanks,

zm711, 2024-07-01:
My hypothesis would be that this is an issue at the [Neo level](https://github.com/NeuralEnsemble/python-neo/), which would need to be implemented there. But could you post the actual error trace so we can see the specific error? Once we know that the error is definitely at the Neo level we will have you open an issue there and if you can provide some data it can be fixed at that level.

h-mayorquin, 2024-07-01:
Which probe are you using, can you share the data?

alejoe91, 2024-07-02:
I think this is still related to https://github.com/NeuralEnsemble/python-neo/issues/1390

Will work on a fix on the NEO side today

alejoe91, 2024-07-04:
@twheatcroft07 can you try to install NEO from this PR?
https://github.com/NeuralEnsemble/python-neo/pull/1501

zm711, 2024-07-10:
@twheatcroft07 the PR has been merged into main at neo. Do you want to test this out and let us know if installing neo from source has fixed your issue?

twheatcroft07, 2024-12-05:
Sorry @zm711 our pipeline has changed so I haven't tested this

zm711, 2024-12-05:
@alejoe91 you want to leave this open or close this since we likely won't get clarity here?

And thanks @twheatcroft07! No worries, totally understand pipelines changing! :)

h-mayorquin, 2024-12-10:
Let's close this.  No trace, no data and the user is not longer using this.

closed 2024-12-10 by h-mayorquin