SpikeInterface/spikeinterface #3111: Metadata handling in `.save_to_folder()`

cwindolf, 2024-07-01:
I think this might be more of a question than an issue, but I wanted to ask: how can I save a recording to binary while maintaining important metadata?

Consider two cases:
 - I preprocess a recording, including standardization with `si.zscore()`. This has some important properties I want to keep track of, like the `gain/offset_to_uV` as well as the zscore's `gain` and `offset` properties.
 - I make a hybrid recording with the new `InjectDriftingTemplatesRecording`. I want to keep track of the `.sorting` and the motion properties.

After either of the above, I `rec.save_to_folder()` to cache the preprocessed or injected data for future use. But, when I load up that saved recording, properties are not set anymore: `gain_to_uV` is all 1s, in the first case, and the `.sorting` is gone in the second case. On the other hand, these properties can be maintained by dumping to pickle, but then I can't save the modified traces to a cached binary file.

Is (or should) there be a way to save metadata and modified traces at once?

alejoe91, 2024-07-01:
Hi @cwindolf 

Labeled as "question" indeed ;)

You can use the `save_to_folder` (or just `save`) to have a quick-access binary version of the data (helpful for hybrid data because it might be slowish to compute) and reload the original non-binary version with:

```
recording_prov = si.load_extractor(saved_folder / "provenance.json")
```

cwindolf, 2024-07-01:
Thanks! Had no idea! That's great.

closed 2024-07-01 by cwindolf

samuelgarcia, 2024-07-03:
The gain/offset are saved in the binary format.
After zscore the new gain is 1 always!
The gain/offset is the one of the latest child in the preprocessing chain.
So you have the real gain when reading a format (spikeglx) filtering do not change it bu whitening or zscore do change it.

cwindolf, 2024-07-03:
I see, thanks Sam! Yeah, I sorta think that the gain_to_uV after zscore should be gain_to_uV/gain, but it's fine because I can work around this by reconstructing from the provenance file.