SpikeInterface/spikeinterface #3112: Getting spike interface QC metrics into Phy

twheatcroft07, 2024-07-01:
Hi there,

I am wondering how to view the quality metrics, computed by spike interface, in Phy. Ideally for each unit in Phy, I would be able to see its computer quality metric scores.

I understand that one can create a sorting analyzer, and compute certain values within that, but I can't see how to associate metrics with a sorting analyzer object, for example.

Thanks,

jakeswann1, 2024-07-01:
An easy way to do this is to extract the quality metrics as a DataFrame, then save this to a `cluster_something.tsv` file. Assuming you alreay have a SortingAnalyzer with metrics computed, should be able to do this as follows:

```
import pandas as pd

# Create sorting analyzer here

quality_metrics = analyzer.extensions['quality_metrics'].data["metrics"]
quality_metrics.to_csv(phy_folder / 'cluster_metrics.tsv', sep='\t')
quality_metrics.index.name = 'cluster_id'
```

By default, phy should load columns from any .tsv files it finds in the folder named `cluster_xxxx.tsv`, with `cluster_id` as an index column. One thing to note with this approach is that these are not computed dynamically, so if you split or merge units the values won't update for any new cluster IDs. 

Hope this helps!

alejoe91, 2024-07-02:
@twheatcroft07 actually if the `SortingAnalyzer` has the `quality_metrics` extension, each metric should alreayd be saved as a TSV file and available in Phy:

See [this section](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/exporters/to_phy.py#L247-L255)

In this PR, I added more control also to be able to add any custom property attached to the `analyzer.sorting` object: https://github.com/SpikeInterface/spikeinterface/pull/3123

twheatcroft07, 2024-07-09:
@alejoe91, thanks so much for working on this. 

As part of export_to_phy(), fitting PCA was slated to take ~3hrs for my example session, so I'll go with @jakeswann1 's option for the time being.

PR #3123 (2024-07-03): Improve export to Phy property handling
Fixes #3112