SpikeInterface/spikeinterface #3113: Add a `display_provenance()` function (or `.provenance`)

alejoe91, 2024-07-01:
We have now beautiful `html` repr of SI objects (thansk @h-mayorquin )

It would be nice to have a similar view for provenance, basically showing an interactive version of the `to_dict` in a notebook. This would also make it easy to navigate through parent objects

h-mayorquin, 2024-07-01:
I like the idea of having the `.provenance`  attribute with an html repr.

zm711, 2024-07-01:
+1 to this!

samuelgarcia, 2024-07-03:
I would like to discuss this before starting.
The provenance should not be loaded when not necessary but only ondemand.
I really want to avoid that the init load the provenance every time for instance

h-mayorquin, 2024-07-03:
I agree that is a good requirement, @samuelgarcia. 

I think that we should stick to the principle of trying to avoid loading at `__init__` in general.