SpikeInterface/spikeinterface #3122: Introduce `codespell` to linting`

JoeZiminski, 2024-07-02:
As someone with absolutely atrocious spelling and propensity to make typos, I have found the linter [codespell](https://github.com/codespell-project/codespell) to be a useful addition to the pre-commit. It basically highlights spelling errors to be fixed before pushing. It is quite light-touch and works well for scientific packages, I've been using it for a long time now and never come across an instance where it is asking me to fix the spelling on a correctly spelled word. Would people be interested?

h-mayorquin, 2024-07-02:
Can it work only in newly introduced changes? If so, I guess that makes sense. + 1

JoeZiminski, 2024-07-02:
I think the CI pre-commit would fail if there are existing spelling errors in the codebase, I think the PR that implements this will have to fix all spelling errors in the codebase (happy to do this). But otherwise do you think this could cause problems?

zm711, 2024-07-02:
I think that PR will be close at 0900 French time.

h-mayorquin, 2024-07-02:
There is one version that works on PRs only without having to be a pre-commit.

samuelgarcia, 2024-07-03:
Une conspiration à nouveau.

zm711, 2024-12-03:
Do we want to continue this discussion or close this as antithetical to Sam?

zm711, 2025-02-21:
I'll close for lack of interest for now. But let the record show I am willing to support this!

closed 2025-02-21 by zm711