SpikeInterface/spikeinterface #3136: Error running read_sorter_folder after installing pre-release

twheatcroft07, 2024-07-03:
Hi there,

I want to use the SortingAnalyzer, so I created a new environment, and installed spikeinterface like so (from [here](https://github.com/SpikeInterface/spikeinterface/issues/2638#issuecomment-2162094216))

pip install --pre spikeinterface

When I try to run the below code I get an error. This code worked fine in the normal version of spikeinterface.

sorting_KS = si.read_sorter_folder(kilosort_output_folder, register_recording=False, sorting_info=False)

The error is below

THANKS!

{
	"name": "UndefinedVariableError",
	"message": "name 'np' is not defined",
	"stack": "---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
File c:\\Users\\sabatini\\.conda\\envs\\twSI\\Lib\\site-packages\\pandas\\core\\computation\\scope.py:231, in Scope.resolve(self, key, is_local)
    230 if self.has_resolvers:
--> 231     return self.resolvers[key]
    233 # if we're here that means that we have no locals and we also have
    234 # no resolvers

File c:\\Users\\sabatini\\.conda\\envs\\twSI\\Lib\\collections\\__init__.py:1015, in ChainMap.__getitem__(self, key)
   1014         pass
-> 1015 return self.__missing__(key)

File c:\\Users\\sabatini\\.conda\\envs\\twSI\\Lib\\collections\\__init__.py:1007, in ChainMap.__missing__(self, key)
   1006 def __missing__(self, key):
-> 1007     raise KeyError(key)

KeyError: 'np'

During handling of the above exception, another exception occurred:

KeyError                                  Traceback (most recent call last)
File c:\\Users\\sabatini\\.conda\\envs\\twSI\\Lib\\site-packages\\pandas\\core\\computation\\scope.py:242, in Scope.resolve(self, key, is_local)
    238 try:
    239     # last ditch effort we look in temporaries
    240     # these are created when parsing indexing expressions
    241     # e.g., df[df > 0]
--> 242     return self.temps[key]
    243 except KeyError as err:

KeyError: 'np'

The above exception was the direct cause of the following exception:

UndefinedVariableError                    Traceback (most recent call last)
Cell In[6], line 55
     52     print(f'KS2.5 sorting completed for {local_path}')
     53 else:
     54     # Read the existing sorter folder
---> 55     sorting_KS = si.read_sorter_folder(kilosort_output_folder, register_recording=False, sorting_info=False)
     56     print(f'KS2.5 folder already exists for {local_path}, skipping spike sorting.')
     58 if run_ironclust:
     59     # Define the IronClust output folder

File c:\\Users\\sabatini\\.conda\\envs\\twSI\\Lib\\site-packages\\spikeinterface\\sorters\\runsorter.py:674, in read_sorter_folder(folder, register_recording, sorting_info, raise_error)
    672 sorter_name = log[\"sorter_name\"]
    673 SorterClass = sorter_dict[sorter_name]
--> 674 sorting = SorterClass.get_result_from_folder(
    675     folder, register_recording=register_recording, sorting_info=sorting_info
    676 )
    677 return sorting

File c:\\Users\\sabatini\\.conda\\envs\\twSI\\Lib\\site-packages\\spikeinterface\\sorters\\basesorter.py:318, in BaseSorter.get_result_from_folder(cls, output_folder, register_recording, sorting_info)
    312     raise SpikeSortingError(
[... 166 lines cut ...]
    544 def visit_Name(self, node, **kwargs) -> Term:
--> 545     return self.term_type(node.id, self.env, **kwargs)

File c:\\Users\\sabatini\\.conda\\envs\\twSI\\Lib\\site-packages\\pandas\\core\\computation\\ops.py:91, in Term.__init__(self, name, env, side, encoding)
     89 tname = str(name)
     90 self.is_local = tname.startswith(LOCAL_TAG) or tname in DEFAULT_GLOBALS
---> 91 self._value = self._resolve_name()
     92 self.encoding = encoding

File c:\\Users\\sabatini\\.conda\\envs\\twSI\\Lib\\site-packages\\pandas\\core\\computation\\ops.py:115, in Term._resolve_name(self)
    110 if local_name in self.env.scope and isinstance(
    111     self.env.scope[local_name], type
    112 ):
    113     is_local = False
--> 115 res = self.env.resolve(local_name, is_local=is_local)
    116 self.update(res)
    118 if hasattr(res, \"ndim\") and res.ndim > 2:

File c:\\Users\\sabatini\\.conda\\envs\\twSI\\Lib\\site-packages\\pandas\\core\\computation\\scope.py:244, in Scope.resolve(self, key, is_local)
    242     return self.temps[key]
    243 except KeyError as err:
--> 244     raise UndefinedVariableError(key, is_local) from err

UndefinedVariableError: name 'np' is not defined"
}

alejoe91, 2024-07-04:
Hi @twheatcroft07 

Seems like a pandas issue? What numpy/pandas versions you have installed? 

We are forcing numpy < 2, and maybe your pandas version is too recent

Jianfengliu0413, 2024-07-09:
it seems "spikeinterface_log.json" is not supported at all, instead, si_folder.json and structures have changed.

alejoe91, 2024-07-09:
right, if this was saved from an old version this might be the reason!

@twheatcroft07 you can always load the KS folder directly (should be in `kilosort_output_folder/sorter_output`) with:
```
sorting=  se.read_kilosort(kilosort_output_folder/sorter_output)
```

zm711, 2024-12-03:
Have this solution worked for you @twheatcroft07 ? Or can we help more?

zm711, 2025-02-21:
I'll close this as stale since we haven't heard back for three months. Feel free to open more issues if anything else comes up!

closed 2025-02-21 by zm711