SpikeInterface/spikeinterface #3144: What module in SI is currently used for cross-session comparison?

taningh86, 2024-07-04:
Hi, 
Since spiketoolkit was deprecated few years ago, what module is used for comparison across different recording sessions?

Thanks

alejoe91, 2024-07-04:
Comparison module ;)

taningh86, 2024-07-04:
Ah ok. Thanks!

taningh86, 2024-07-05:
I get the following error when I try to plot agreement matrix `sw.plot_agreement_matrix(m_tcmp)`. However it works for 'p_tcmp'. Should I be using a different function for 'm_tcmp'?
`AttributeError                            Traceback (most recent call last)
Cell In[10], line 4
      1 import spikeinterface.qualitymetrics as sqm
      2 import spikeinterface.widgets as sw
----> 4 sw.plot_agreement_matrix(m_tcmp)

File ~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\widgets\comparison.py:110, in AgreementMatrixWidget.__init__(self, sorting_comparison, ordered, count_text, unit_ticks, backend, **backend_kwargs)
    101 def __init__(
    102     self, sorting_comparison, ordered=True, count_text=True, unit_ticks=True, backend=None, **backend_kwargs
    103 ):
    104     plot_data = dict(
    105         sorting_comparison=sorting_comparison,
    106         ordered=ordered,
    107         count_text=count_text,
    108         unit_ticks=unit_ticks,
    109     )
--> 110     BaseWidget.__init__(self, plot_data, backend=backend, **backend_kwargs)

File ~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\widgets\base.py:82, in BaseWidget.__init__(self, data_plot, backend, immediate_plot, **backend_kwargs)
     79 self.backend_kwargs = backend_kwargs_
     81 if immediate_plot:
---> 82     self.do_plot()

File ~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\widgets\base.py:103, in BaseWidget.do_plot(self)
    101 def do_plot(self):
    102     func = getattr(self, f"plot_{self.backend}")
--> 103     func(self.data_plot, **self.backend_kwargs)

File ~\anaconda3\envs\si_env\lib\site-packages\spikeinterface\widgets\comparison.py:123, in AgreementMatrixWidget.plot_matplotlib(self, data_plot, **backend_kwargs)
    120 comp = dp.sorting_comparison
    122 if dp.ordered:
--> 123     scores = comp.get_ordered_agreement_scores()
    124 else:
    125     scores = comp.agreement_scores

AttributeError: 'MultiTemplateComparison' object has no attribute 'get_ordered_agreement_scores'`

alejoe91, 2024-07-05:
no, multi template comparison is not as supported as multi sorting comparison. The widgets are only available for the `MultiSortingComparison`

taningh86, 2024-07-05:
Ok. So I am assuming I can just do `p_tcmp` for recording sessions and repeat that for different sessions, that way I cover all the sessions. Let me know if there's a better way.
The other way that comes to mind is to treat different recording sessions I have (all from one sorter) as recordings from different sorters and use `MultiSortingComparison`?

h-mayorquin, 2024-07-10:
Hi, @taningh86, as the question of the issue was answered and you seem to have a way of handling your secondary question I am closing this. Feel free to open as many issues as you like : )

closed 2024-07-10 by h-mayorquin

taningh86, 2024-07-10:
@h-mayorquin for sure. Thank you!