SpikeInterface/spikeinterface #3145: Missing channels after running correct_motion

scg63, 2024-07-04:
Hello! I am preprocessing and doing motion correction on some neuropixel recording data using Spike Interface before I run it using Kilosort 4. Eventually we will run Kilosort 4 directly using Spike Interface but in the meantime, I'm exporting the preprocessed data as a binary file so I can plug it into Kilosort. After running correct_motion, the number of channels decreases from 384 to 382. Has this happened to anyone else? This happens directly after running correct_motion, not after using bandpass_filter/common_reference. Any help or suggestions would be greatly appreciated!

```python
from pathlib import Path
import matplotlib.pyplot as plt
import numpy as np
import shutil
import spikeinterface.full as si
```

```python
raw_rec = si.read_spikeglx(folder_path='D:\data\...\recording_g01_t0.imec0.ap')
raw_rec
```

    SpikeGLXRecordingExtractor: 384 channels - 30.0kHz - 1 segments - 4,410,582 samples 
                                147.02s (2.45 minutes) - int16 dtype - 3.15 GiB

```python
def preprocess_chain(rec):
	rec = si.bandpass_filter(rec, freq_min=300., freq_max=6000.)
	rec = si.common_reference(rec, reference='global', operator = 'median')
	return rec

rec = preprocess_chain(raw_rec)
```

```python
global_job_kwargs = dict(n_jobs=40, chunk_duration='1s', progress_bar=True)
si.set_global_job_kwargs(**global_job_kwargs)
```

```python
motion_folder = 'D:\data\...\recording_g01_t0.imec0.ap\\resultsSI'
rec_corrected, motion_info = si.correct_motion(rec, preset='nonrigid_accurate', folder=motion_folder, output_motion_info=True)
```

    detect and localize:   0%|          | 0/148 [00:00<?, ?it/s]

```python
rec_corrected
```

    InterpolateMotionRecording: 382 channels - 30.0kHz - 1 segments - 4,410,582 samples 
                                147.02s (2.45 minutes) - int16 dtype - 3.14 GiB

```python
rec_corrected = si.scale(rec_corrected, gain=200)
rec_corrected = rec_corrected.astype("int16")
rec_corrected = rec_corrected.save(folder='D:\data\...\recording_g01_t0.imec0.ap\\resultsSI\preprocess', n_jobs=1,format='binary')
```

    write_binary_recording with n_jobs = 1 and chunk_size = 29999

    write_binary_recording:   0%|          | 0/148 [00:00<?, ?it/s]

```python
rec_corrected

```

    BinaryFolderRecording: 382 channels - 30.0kHz - 1 segments - 4,410,582 samples 
                           147.02s (2.45 minutes) - int16 dtype - 3.14 GiB

```python

```

yger, 2024-07-04:
This is a normal behaviour, since by default motion correction do not extrapolate for channels at the border, and only consider the ones that will be within the reach of the estimated motion. You should have a look to the documentation, and play with the interpolate_motion_kwargs parameters, setting border_mode to "for e_extrapolate" instead of "remove_channels"

scg63, 2024-07-05:
@yger Thank you for explaining that! I had missed that in the documentation. I'll play around with those parameters.

ecobost, 2026-03-04:
@yger do you wanna close this?