SpikeInterface/spikeinterface #3146: "Can't interpolate traces" with int16 dtype in motion_correction

avakiai, 2024-07-04:
Hello, 

I'm trying to try out a few motion correction algorithms on Neuropixel 1.0 data. I'm currently running through the pipeline with a short (<5 minute) recording. After chaining preprocessing steps, I get the following error when the detect and localize step completes. Any ideas what is causing this behavior? 

At first I thought the error may have been caused by one bad channel that had been removed, but the error persists regardless of whether any channels are removed or not. 
Also, the dtype has not changed through preprocessing - og dtype is and remains int16.

```
job_kwargs = dict(n_jobs=40, chunk_duration='1s', progress_bar=True)

rec1 = si.highpass_filter(raw_rec, freq_min=400.)
bad_channel_ids, channel_labels = si.detect_bad_channels(rec1)
rec2 = rec1.remove_channels(bad_channel_ids)
print('bad_channel_ids', bad_channel_ids)

rec3 = si.phase_shift(rec1)
rec4 = si.common_reference(rec3, operator="median", reference="global")
rec = rec4

some_presets = ('kilosort_like', 'nonrigid_accurate')
for preset in some_presets:
    print('Computing with', preset)
    folder = anim_dat_dir[2] + '/' + 'motion_folder_dataset1' + '/' + preset

    # if folder.exists():
    #     shutil.rmtree(folder)
    recording_corrected, motion_info = si.correct_motion(rec, preset=preset,
                                                         folder=folder,
                                                         output_motion_info=True, **job_kwargs)

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[57], line 6
      3 folder = anim_dat_dir[2] + '/' + 'motion_folder_dataset1' + '/' + preset
      4 # if folder.exists():
      5 #     shutil.rmtree(folder)
----> 6 recording_corrected, motion_info = si.correct_motion(rec, preset=preset,
      7                                                      folder=folder,
      8                                                      output_motion_info=True, **job_kwargs)

File D:\downloads_\spikeinterface\src\spikeinterface\preprocessing\motion.py:386, in correct_motion(recording, preset, folder, output_motion_info, overwrite, detect_kwargs, select_kwargs, localize_peaks_kwargs, estimate_motion_kwargs, interpolate_motion_kwargs, **job_kwargs)
    383 t1 = time.perf_counter()
    384 run_times["estimate_motion"] = t1 - t0
--> 386 recording_corrected = InterpolateMotionRecording(recording, motion, **interpolate_motion_kwargs)
    388 motion_info = dict(
    389     parameters=parameters,
    390     run_times=run_times,
   (...)
    393     motion=motion,
    394 )
    395 if folder is not None:

File D:\downloads_\spikeinterface\src\spikeinterface\sortingcomponents\motion_interpolation.py:343, in InterpolateMotionRecording.__init__(self, recording, motion, border_mode, spatial_interpolation_method, sigma_um, p, num_closest, interpolation_time_bin_centers_s, interpolation_time_bin_size_s, dtype, **spatial_interpolation_kwargs)
    341         dtype = recording.dtype
    342     else:
--> 343         raise ValueError(f"Can't interpolate traces of recording with non-floating dtype={recording.dtype=}.")
    345 dtype_ = fix_dtype(recording, dtype)
    346 BasePreprocessor.__init__(self, recording, channel_ids=channel_ids, dtype=dtype_)

ValueError: Can't interpolate traces of recording with non-floating dtype=recording.dtype=dtype('int16').

```

Many thanks,
ava

avakiai, 2024-07-04:
Update: changing dtype to float coming out of preprocessing chain solved the issue and correct_motion runs with both presets now. (Is there any reason why this would alter spike sorting outcomes, and should be changed back to original int16 at some point?)

However, trying to plot the motion vector as per the documentation now gives an error that the API has been updated...
```
for preset in some_presets:
    # load
    folder = anim_dat_dir[2] + '/' + 'motion_folder_dataset1' + '/' + preset
    motion_info = si.load_motion_info(folder)

    # and plot
    fig = plt.figure(figsize=(14, 8))
    si.plot_motion(motion_info, figure=fig, depth_lim=(400, 600),
                   color_amplitude=True, amplitude_cmap='inferno', scatter_decimate=10)
    fig.suptitle(f"{preset=}")
    
  ---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[90], line 8
      6 # and plot
      7 fig = plt.figure(figsize=(14, 8))
----> 8 si.plot_motion(motion_info, figure=fig, depth_lim=(400, 600),
      9                color_amplitude=True, amplitude_cmap='inferno', scatter_decimate=10)
     10 fig.suptitle(f"{preset=}")

File D:\downloads_\spikeinterface\src\spikeinterface\widgets\motion.py:38, in MotionWidget.__init__(self, motion, segment_index, mode, motion_lim, backend, **backend_kwargs)
     28 def __init__(
     29     self,
     30     motion: Motion,
   (...)
     35     **backend_kwargs,
     36 ):
     37     if isinstance(motion, dict):
---> 38         raise ValueError(
     39             "The API has changed, plot_motion() used Motion object now, maybe you want plot_motion_info(motion_info)"
     40         )
     42     if segment_index is None:
     43         if len(motion.displacement) == 1:

ValueError: The API has changed, plot_motion() used Motion object now, maybe you want plot_motion_info(motion_info)
 
```
... and then when run with the current function call (`plot_motion_info`), throws an error several layers deep in the callback. Perhaps a bug? 

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[91], line 8
      6 # and plot
      7 fig = plt.figure(figsize=(14, 8))
----> 8 si.plot_motion_info(motion_info, figure=fig, depth_lim=(400, 600),
      9                color_amplitude=True, amplitude_cmap='inferno', scatter_decimate=10)
     10 fig.suptitle(f"{preset=}")

File D:\downloads_\spikeinterface\src\spikeinterface\widgets\motion.py:364, in MotionInfoWidget.__init__(self, motion_info, segment_index, recording, depth_lim, motion_lim, color_amplitude, scatter_decimate, amplitude_cmap, amplitude_color, amplitude_clim, amplitude_alpha, backend, **backend_kwargs)
    346 times = recording.get_times() if recording is not None else None
    348 plot_data = dict(
    349     sampling_frequency=motion_info["parameters"]["sampling_frequency"],
    350     times=times,
   (...)
    361     **motion_info,
    362 )
--> 364 BaseWidget.__init__(self, plot_data, backend=backend, **backend_kwargs)

File D:\downloads_\spikeinterface\src\spikeinterface\widgets\base.py:87, in BaseWidget.__init__(self, data_plot, backend, immediate_plot, **backend_kwargs)
     84 self.backend_kwargs = backend_kwargs_
     86 if immediate_plot:
---> 87     self.do_plot()

File D:\downloads_\spikeinterface\src\spikeinterface\widgets\base.py:108, in BaseWidget.do_plot(self)
    106 def do_plot(self):
    107     func = getattr(self, f"plot_{self.backend}")
--> 108     func(self.data_plot, **self.backend_kwargs)

File D:\downloads_\spikeinterface\src\spikeinterface\widgets\motion.py:404, in MotionInfoWidget.plot_matplotlib(self, data_plot, **backend_kwargs)
    401 ax1.sharex(ax0)
    402 ax1.sharey(ax0)
--> 404 corrected_location = correct_motion_on_peaks(
    405     dp.peaks,
    406     dp.peak_locations,
    407     dp.motion,
    408     dp.recording,
    409 )
    411 commpon_drift_map_kwargs = dict(
    412     direction=dp.motion.direction,
    413     recording=dp.recording,
   (...)
    422     backend="matplotlib",
    423 )
    425 # with immediate_plot=True the widgets are plotted immediately

File D:\downloads_\spikeinterface\src\spikeinterface\sortingcomponents\motion_interpolation.py:33, in correct_motion_on_peaks(peaks, peak_locations, motion, recording)
     30 corrected_peak_locations = peak_locations.copy()
     32 for segment_index in range(motion.num_segments):
---> 33     times_s = recording.sample_index_to_time(peaks["sample_index"], segment_index=segment_index)
     34     i0, i1 = np.searchsorted(peaks["segment_index"], [segment_index, segment_index + 1])
     36     spike_times = times_s[i0:i1]

AttributeError: 'NoneType' object has no attribute 'sample_index_to_time'
```

samuelgarcia, 2024-07-04:
Hi Ava.
I guess that you are working with the main branch
The motion estimation is totally work-in-progress theses times
Doc has not been updated yet.
when this #3062 will be merged (new week maybe) you could have even more changes. (but new preset like dredge!)
sorry for that.
A new doc should availble when merging.

And yes the solution was to dtype as float32 like you said.

For plotting you can do this
```python
motion_info = si.load_motion_info(folder)
motion = motion_info['motion']  #<<< this is the Motion object
print(motion)
plot_motion(motion, mode='map')
plot_motion(motion, mode='line')
```

zm711, 2024-07-19:
I'll close this but if something else comes up let us know!

closed 2024-07-19 by zm711