SpikeInterface/spikeinterface #3149: Extracting quality metrics with or without a sortingAnalyzer

twheatcroft07, 2024-07-04:
Hi there,

Originally I was computing metrics via extract_waveforms then compute_quality_metrics(we, metric_names=['amplitude_median', 'sliding_rp_violation', 'amplitude_cutoff']) , from [this page](https://spikeinterface.readthedocs.io/en/0.91.0/modules/toolkit/plot_3_quality_mertics.html) . I understand this process is deprecated now.

This would take 30mins per session. 

I want to get the metrics into Phy, and I was suggested to create a sorting analyzer instead ([here](https://github.com/SpikeInterface/spikeinterface/issues/3112)). To get access to sorting analyzers, I installed spikeinterface 'from source' (see [here](https://spikeinterface.readthedocs.io/en/stable/installation.html#from-source)).

To extract the metrics via the analyzer, I ran ->
 
 `analyzer = si.create_sorting_analyzer(sorting_KS, rec, sparse=True,  format="binary_folder", folder=analyzerPath)`
 `analyzer.compute("random_spikes")`
`analyzer.compute("waveforms")`
`analyzer.compute("templates")`
`analyzer.compute("noise_levels")`
`qm_ext = analyzer.compute(input="quality_metrics", metric_names=['amplitude_cutoff'], skip_pc_metrics=True)`

This took more like 3.5 per session, with the first and third lines taking 1.25 and 2.25hrs respectively. Moreover, asking the sorting analyzer to compute 'sliding_rp_violations' gives an error.

Is this expected? 

If people would prefer to use the quicker metric extraction route (i.e. not using sorting analyzer), is there still a way of getting those metrics into Phy?

Thanks,

samuelgarcia, 2024-07-05:
Hi.
thanks for the feedback.
the new sortinganalyzer is supposed to be faster more flexible than the old waveform extractor.
Here in your case apparently not! I am sorry for that. But this is strange. Did you do n_jobs=-1 in the job kwargs ?

One of my guess is that you an enormous units count.
Could you give some number ? channels, units counts, duration ? where is located the recording (ssd, network drive, ...)  ?

The first step (create_sorting_analyzer) estimate the sparsity automatically and need to go to the entire traces, this can be slow in some cases.
The third step (waveforms) can now be skip unless you need pca

Can you copy the error you have with sliding_rp_violations ?

alejoe91, 2024-07-05:
@samuelgarcia my guess is that it's running with 1 job!

twheatcroft07, 2024-07-05:
Hi both,

I really appreciate all the support offered. 

- The recording and the sorting are both stored locally, on an SSD. 
- My recording is 50.3 mins, at 30kHz, with 384 channels (NP2.0). I only have 222 units.

It appears that the issue was with n_jobs=-1 in the job kwargs. I didn't have any kwargs before. Without n_jobs, I can reproduce the durations from before (see screenshots below). With n_jobs = -1, I can get the time back to what it was with the waveform extractor. 

[image]
[image]

The n_jobs input is suggested [here](https://github.com/SpikeInterface/spikeinterface-gui?tab=readme-ov-file), which I just came across, but perhaps it could be suggested on the API examples [here](https://spikeinterface.readthedocs.io/en/latest/api.html#spikeinterface.core.create_sorting_analyzer) and on the postprocessing page [here](https://spikeinterface.readthedocs.io/en/latest/modules/postprocessing.html#computing-extensions) since it does seem to be a game-changer in terms of speed, as you suggested.

I can no longer reproduce my sliding_rp_violation issue, so perhaps I did something weird before, apologies!

Thanks again!

samuelgarcia, 2024-07-05:
Cool.

zm711, 2024-07-05:
@twheatcroft07 

If you look at your terminal output there is also a `UserWarning` saying to change from n_jobs=1. Is that warning not clear? Is there something that would make it clearer?

twheatcroft07, 2024-07-05:
@zm711 

Very true! I missed that but you're totally right, I suspect people will mostly see that and adjust. My bad.

closed 2024-07-06 by alejoe91