SpikeInterface/spikeinterface #3155: Issue with run_sorter() command

sushmitaS16, 2024-07-05:
Hello,
I am using Spike Interface to analyze open-ephys data and it seems to get stuck at the run_sorter() command.

Exact situation: this command was running for 1 whole week with no further progress! 
[image]

I tried the UserWarning mentioned to parallelize my task, but that too gives an attribute error:
[image]

My system specs are as follows:
Debian GNU/Linux 11 (bullseye)
5.10.0-29-amd64

Also, I am using python v3.9 and spike interface v0.100.8

I would highly appreciate if anyone could help me figure this out!
Thanks

zm711, 2024-07-05:
Do you really have 115 cores? This seems like too many processes to spin off, which to my mind could explain part of the problem. Are you on a cluster? 

As for your immediate error you should do the following instead;

```python
si.set_global_job_kwargs(**job_kwargs)
```

We've made some updates in 0.101.0 but I forget which ones made it into 0.100.8.

h-mayorquin, 2024-07-05:
Hi, for the second error, run_sorter for spykingcircus takes the job_kwargs like this:

```
run_sorter("spykingcircus2", recording=recording, job_kwargs={"n_jobs":3})
```

As a parameter and not as kwargs

sushmitaS16, 2024-07-09:
Hello,
@zm711 Yes, I am running this code in a cluster with 144 cores. I tried the way you mentioned but still getting the same error.

@h-mayorquin Thanks for the syntax correction, I'll try that out.
Following are the details that you requested:
1. Cores : 144
2. RAM : 528 GB
3. Recording format: Open-ephys
4. Preprocessing used: 
Bandpass filtering : recording_bp = bandpass_filter(recording=raw_rec, freq_min=500, freq_max=5000) 
Common Referencing : recording_cmr = common_reference(recording=recording_bp, operator='median')
HighAPI for motion correction : recording_mc = correct_motion(recording=recording_cmr, preset='rigid_fast')

Thanks

h-mayorquin, 2024-07-12:
Let us know how it goes. Is the thing still hanging?

sushmitaS16, 2024-09-20:
Hello, 
spykingcircus2 never worked for us!
We decided to switch to Kilosort4 and it worked fine.

Thanks

closed 2024-09-20 by sushmitaS16