SpikeInterface/spikeinterface #3166: Bug

BEGINRX, 2024-07-09:


yger, 2024-07-09:
You can try to switch to the circus-omp-svd method for now, in the merging parameters of spyking-circus2, this should work. I'll try to see what is the bug with wobble

BEGINRX, 2024-07-10:
Thanks for reply. I have used the circus-omp-svd method to sort. But there also is an error.
[image]

Is this because there is something wrong with my data？Would that happen if there were very few spikes in the signal?

BEGINRX, 2024-07-10:
But there are many spikes detected... 
[image]

PR #3178 (2024-07-15): Fix pca transform error
I have actually encountered #3166 a few times, and it should be fixed with this simple solution. We were a bit too strict with the exception type.

Fixes #3166
closed 2024-07-15 by samuelgarcia