SpikeInterface/spikeinterface #3175: Stuck process in Motion_correction

florgf88, 2024-07-10:
Hi! I've been trying the drift correction module recently, and I found it to be a really nice combination of tools for addressing this issue. However, I'm having trouble getting the final output from motion_correction: it completes the peak detection and localization (shown in the progress bar) and then gets stuck there for several hours without completing the process. I would like to save this output to run kilosort on the corrected data. I tried so far to run this from a jupyter notebook, using spike interface 0.100.8 and 0.101 in Windows 11.

Any suggestions regarding this are very much appreciated, many thanks!

[image]
[image]
[image]

JoeZiminski, 2024-07-10:
Hi @florgf88, the `nonrigid_accurate` method does take quite a long time, and I have also noticed it is confusing as nothing is printed when it begins, it would be useful to print something here.

I think it is probably running, and just taking a while, how large is your recording? It might also be worth trying "nonrigid_fast_and_accurate" which is a similar method with some speedups applied. @samuelgarcia @cwindolf will be able to inform further on the specific details there.

florgf88, 2024-07-10:
Many thanks @JoeZiminski for the help! My recording is 2.5 hours long and was performed with Nrpx 1.0. I can see first if it completes overnight before trying "nonrigid_fast_and_accurate".

florgf88, 2024-07-11:
Hi, the process took around 13 hours to complete, but finished and saved the output correctly. I agree that a progress bar or a printout would be useful. I will go ahead with "nonrigid_fast_and_accurate" and see how it goes. Many thanks for the help!