SpikeInterface/spikeinterface #3181: Lots of sphinx-build warnings

chrishalcrow, 2024-07-11:
When you build the docs using sphinx, there are lots of warnings. I had a go at getting rid of them, and came across something that needs discussion:

About 120/150 of the warnings are to do with repeated headings. Sphinx's `autosectionlabel` extension can use titles of pages, sections and subsections as references. So if you write something like `:ref:'Run motion correction with one function!'` it will create a link to [the page containing this section heading](https://spikeinterface.readthedocs.io/en/latest/how_to/handle_drift.html#run-motion-correction-with-one-function).

However, many headings are generic. E.g. every quality metric has a `Calculation` and an `Expectations and use` heading (https://spikeinterface.readthedocs.io/en/latest/modules/qualitymetrics/isolation_distance.html). And each of these raise (lots!) of warnings.

Two ways I see to solve this:
1) Make all section headings unique (e.g. use `Isolation distance calculation` and `Isolation distance expectation/use`)
Pro: we can uniquely refer to any heading. Cool!
Con: Pages like the [Isolation distance](https://spikeinterface.readthedocs.io/en/latest/how_to/handle_drift.html#run-motion-correction-with-one-function) will get a bit repetitive. 
2) Turn off this warning by adding `suppress_warnings = ['autosectionlabel.*']` to `conf.py`. 
Pro: We can repeat headings
Con: We might get unexpected behaviour if we don't realise there are repeated headings. E.g. at the moment if we link to SortingAnalyzer sphinx will pick either [this page](https://spikeinterface.readthedocs.io/en/latest/tutorials/core/plot_4_sorting_analyzer.html#sortinganalyzer) or [this page](https://spikeinterface.readthedocs.io/en/latest/tutorials/core/plot_4_sorting_analyzer.html#sortinganalyzer), essentially at random.

Thoughts?

zm711, 2024-07-11:
That's super tough from my perspective. I think needing to change section heading for each quality metric would be annoying, but not the end of the world. I'm super bad about doing the linking so I personally never notice the con of 2. But I can see how that would be a big problem. 

I think if I were forced to vote I would go for 1 to make explicit and reduce bugs in our docs/rendering/linking even if I think it is a little uglier/repetitive.

JoeZiminski, 2024-07-11:
Thanks for looking into this those warnings were super annoying / mysterious! I agree with Zach it's not an easy fix, I would probably lean towards supressing the warnings it will be strange to read, and ideally the behind-the-scenes of sphinx should not dictate how the documentation is presented. We can always use [explicit targets](https://docs.readthedocs.io/en/stable/guides/cross-referencing-with-sphinx.html#explicit-targets) to target these headers. 

It's a good point though if somone tries to use the other linking method to a subsection but does not realise the subsection is duplicated. I think however this should raise a separate warning, that the link does not point to a unique header. Would you mind trying this out? (i.e. supress `autoselectionlabel.*` but then create a link to a duplicated header and see if it raises a separate warning).

chrishalcrow, 2024-07-11:
> It's a good point though if somone tries to use the other linking method to a subsection but does not realise the subsection is duplicated. I think however this should raise a separate warning, that [...]

Sadly not: the warning for duplicated section headings is the same as the warnings for when you try to link to a duplicated section heading. So it's all warnings, or no warnings.

Another options: the `autosectionlabel_prefix_document` option means we have to specify the filename the header is in. So you'd specify e.g. :ref:`modules/core:SortingAnalyzer` or :ref:`tutorials/core/plot_4_sorting_analyzer:SortingAnalyzer`. Breaks some of the sphinx magic (since you need to know where the files are) but we get warnings when things are broken and can have repeated headings. Thoughts? Not a big deal to implement, we have about 30 internal refs.

JoeZiminski, 2024-07-11:
Nice! I think that's my preferred option, like you say it's annoying it will break if files are moved but you get the warning it's broken and that's the main thing. In this case my preference is: your last suggestion > change all headers so they are not duplicated > supress warnings.

h-mayorquin, 2024-07-11:
> Another options: the autosectionlabel_prefix_document option means we have to specify the filename the header is in. So you'd specify e.g. :ref:modules/core:SortingAnalyzer or [...]

I think this is the way to go.
The person referencing should know where is the file of the heading that they want to reference.
Is a higher demand to ask the person who is creating a new heading to know that no heading of equal name has been created.

zm711, 2024-07-11:
> Is a higher demand to ask the person who is creating a new heading to know that no heading of equal name has been created.

I think this is a really really strong point. At this point I don't know if anyone actually knows all headers in the docs (I sure don't). So I don't think we can rely on that. It would be too much.

chrishalcrow, 2024-07-12:
Excellent! That's a unanimous vote! I'll get a PR going...

closed 2024-07-19 by chrishalcrow