SpikeInterface/spikeinterface #3182: Phy TemplateFeatureView: template_feature_ind and templates_features not created

florian6973, 2024-07-11:
Hi!

I was opening a new issue because I noticed that I cannot create a TemplateFeatureView in phy after export from spikeinterface (0.100.8, 0.101.0rc0). As discussed here https://github.com/cortex-lab/phy/issues/1290, it seems that the files template_feature_ind and templates_features are not generated. Is it on purpose, or could we add them by any chance please? Or should we wait for spikeinterface-gui curation features? Thanks!

Best,

zm711, 2024-07-11:
@alejoe91, 

I said I honestly didn't know why we didn't make these files. So I need to leave this one to you to answer.

alirezasaeedi1988, 2024-08-02:
Hey there,

Is there any update on this?

zm711, 2024-09-03:
@alejoe91 and @samuelgarcia wanted to bump this. Is there are design reason not to do this or was it just lack of time?

zm711, 2025-08-21:
This is now stale. We are actively working on the [spikeinterface-gui](https://github.com/SpikeInterface/spikeinterface-gui). So our efforts are currently on providing the best possible integration at that level. I will close this for now.

closed 2025-08-21 by zm711