SpikeInterface/spikeinterface #3183: Kilosort error: The expanded size of the tensor (60122) must match the existing size (60071) at non-singleton dimension 1.  Target sizes: [112, 60122].  Tensor sizes: [112, 60071]

JuanPimientoCaicedo, 2024-07-11:
Hello, everyone. I saw that this error happened before https://github.com/SpikeInterface/spikeinterface/issues/2699. And I tried to use python == 3.9 and CUDA = 11.8.

But I am still getting the same error:

```

INFO:kilosort.io:========================================
INFO:kilosort.io:Loading recording with SpikeInterface...
INFO:kilosort.io:number of samples: 1800010
INFO:kilosort.io:number of channels: 112
INFO:kilosort.io:numbef of segments: 1
INFO:kilosort.io:sampling rate: 30000.18060200669
INFO:kilosort.io:dtype: int16
INFO:kilosort.io:========================================
INFO:kilosort.run_kilosort: 
INFO:kilosort.run_kilosort:Computing preprocessing variables.
INFO:kilosort.run_kilosort:----------------------------------------
INFO:kilosort.run_kilosort:Preprocessing filters computed in  7.90s; total  7.90s
INFO:kilosort.run_kilosort: 
INFO:kilosort.run_kilosort:Computing drift correction.
INFO:kilosort.run_kilosort:----------------------------------------
INFO:kilosort.datashift:nblocks = 0, skipping drift correction
INFO:kilosort.run_kilosort:drift computed in  0.00s; total  7.90s
INFO:kilosort.run_kilosort: 
INFO:kilosort.run_kilosort:Extracting spikes using templates
INFO:kilosort.run_kilosort:----------------------------------------
INFO:kilosort.spikedetect:Re-computing universal templates from data.
Skipping drift correction.
 94%|████████████████████████████████████████████████████████████████████████████▋     | 29/31 [02:19<00:09,  4.81s/it]
Error running kilosort4
---------------------------------------------------------------------------
SpikeSortingError                         Traceback (most recent call last)
Cell In[19], line 3
      1 job_kwargs = dict(n_jobs=-1, chunk_duration='1s', progress_bar=True)
----> 3 sorting = si.run_sorter(sorter_name = 'kilosort4', recording = cortex_AP_slice, 
      4                         folder = Path('C:/Users/Juan/Desktop/KS4_output'), 
      5                         do_correction = False, verbose = True, docker_image = False)

File ~\.conda\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py:216, in run_sorter(sorter_name, recording, folder, remove_existing_folder, delete_output_folder, verbose, raise_error, docker_image, singularity_image, delete_container_files, with_output, output_folder, **sorter_params)
    205             raise RuntimeError(
    206                 "The python `spython` package must be installed to "
    207                 "run singularity. Install with `pip install spython`"
    208             )
    210     return run_sorter_container(
    211         container_image=container_image,
    212         mode=mode,
    213         **common_kwargs,
    214     )
--> 216 return run_sorter_local(**common_kwargs)

File ~\.conda\envs\si_env\lib\site-packages\spikeinterface\sorters\runsorter.py:276, in run_sorter_local(sorter_name, recording, folder, remove_existing_folder, delete_output_folder, verbose, raise_error, with_output, output_folder, **sorter_params)
    274 SorterClass.set_params_to_folder(recording, folder, sorter_params, verbose)
    275 SorterClass.setup_recording(recording, folder, verbose=verbose)
--> 276 SorterClass.run_from_folder(folder, raise_error, verbose)
    277 if with_output:
    278     sorting = SorterClass.get_result_from_folder(folder, register_recording=True, sorting_info=True)

File ~\.conda\envs\si_env\lib\site-packages\spikeinterface\sorters\basesorter.py:301, in BaseSorter.run_from_folder(cls, output_folder, raise_error, verbose)
    298         print(f"{sorter_name} run time {run_time:0.2f}s")
    300 if has_error and raise_error:
--> 301     raise SpikeSortingError(
    302         f"Spike sorting error trace:\n{error_log_to_display}\n"
    303         f"Spike sorting failed. You can inspect the runtime trace in {output_folder}/spikeinterface_log.json."
    304     )
    306 return run_time

SpikeSortingError: Spike sorting error trace:
Traceback (most recent call last):
  File "C:\Users\Juan\.conda\envs\si_env\lib\site-packages\spikeinterface\sorters\basesorter.py", line 261, in run_from_folder
    SorterClass._run_from_folder(sorter_output_folder, sorter_params, verbose)
  File "C:\Users\Juan\.conda\envs\si_env\lib\site-packages\spikeinterface\sorters\external\kilosort4.py", line 261, in _run_from_folder
    st, tF, _, _ = detect_spikes(ops, device, bfile, tic0=tic0, progress_bar=progress_bar)
  File "C:\Users\Juan\.conda\envs\si_env\lib\site-packages\kilosort\run_kilosort.py", line 481, in detect_spikes
    st0, tF, ops = spikedetect.run(ops, bfile, device=device, progress_bar=progress_bar)
  File "C:\Users\Juan\.conda\envs\si_env\lib\site-packages\kilosort\spikedetect.py", line 250, in run
    X = bfile.padded_batch_to_torch(ibatch, ops)
  File "C:\Users\Juan\.conda\envs\si_env\lib\site-packages\kilosort\io.py", line 776, in padded_batch_to_torch
    X = super().padded_batch_to_torch(ibatch)
  File "C:\Users\Juan\.conda\envs\si_env\lib\site-packages\kilosort\io.py", line 601, in padded_batch_to_torch
    X[:] = torch.from_numpy(data).to(self.device).float()
RuntimeError: The expanded size of the tensor (60122) must match the existing size (60071) at non-singleton dimension 1.  Target sizes: [112, 60122].  Tensor sizes: [112, 60071]

```

Python version: 3.9.19 | packaged by conda-forge | (main, Mar 20 2024, 12:38:46) [MSC v.1929 64 bit (AMD64)]
CUDA is available:  True - version 11.8
torch version:  2.3.1
si version:  0.101.0rc0
kilosort version:  4.0.12

Do you know what I am doing wrong here?

alejoe91, 2024-07-11:
Hi,

Have you asked on the Kilosort repo? Never seen this before unfortunately

JuanPimientoCaicedo, 2024-07-11:
Yes, I did. It worked perfectly with both 4.0.12 and 4.0.13 versions using Kilosort gui. I know the latest SI update supported 4.0.12, but I never tested myself. I will let you know if I find out what is happening.

JuanPimientoCaicedo, 2024-07-11:
It was a kilosort problem, guys. I could replicate it with the GUI by using the real ADC sampling rate. since in this case, it was 30000.18060200669 Hz, it probably is a rounding problem.

closed 2024-07-11 by JuanPimientoCaicedo