SpikeInterface/spikeinterface #3186: Potential Stochastic Test Failure in `test_generate_sorting_to_inject`

zm711, 2024-07-11:
This only failed on one OS/one python version. Not sure if this is a problem in generate_sorting_to_inject which can cause some failures. We can leave this issue open to monitor if this problem happens multiple times.

From action: https://github.com/SpikeInterface/spikeinterface/actions/runs/9898102287/job/27344129816?pr=3185

```
_______________________ test_generate_sorting_to_inject ________________________

    def test_generate_sorting_to_inject():
        durations = [10.0, 20.0]
        sorting = generate_sorting(num_units=10, durations=durations, sampling_frequency=30000, firing_rates=1.0)
>       injected_sorting = generate_sorting_to_inject(
            sorting, [int(duration * sorting.sampling_frequency) for duration in durations]
        )

src/spikeinterface/core/tests/test_generate.py:549: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
src/spikeinterface/core/generate.py:293: in generate_sorting_to_inject
    min_diff = np.min(np.abs(injected_spike_train[:, None] - spike_train[None, :]), axis=1)
/Library/Frameworks/Python.framework/Versions/3.12/lib/python3.12/site-packages/numpy/core/fromnumeric.py:2953: in min
    return _wrapreduction(a, np.minimum, 'min', axis, None, out,
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

obj = array([], shape=(0, 0), dtype=int64), ufunc = <ufunc 'minimum'>
method = 'min', axis = 1, dtype = None, out = None
kwargs = {'initial': <no value>, 'keepdims': <no value>, 'where': <no value>}
passkwargs = {}

    def _wrapreduction(obj, ufunc, method, axis, dtype, out, **kwargs):
        passkwargs = {k: v for k, v in kwargs.items()
                      if v is not np._NoValue}
    
        if type(obj) is not mu.ndarray:
            try:
                reduction = getattr(obj, method)
            except AttributeError:
                pass
            else:
                # This branch is needed for reductions like any which don't
                # support a dtype.
                if dtype is not None:
                    return reduction(axis=axis, dtype=dtype, out=out, **passkwargs)
                else:
                    return reduction(axis=axis, out=out, **passkwargs)
    
>       return ufunc.reduce(obj, axis, dtype, out, **passkwargs)
E       ValueError: zero-size array to reduction operation minimum which has no identity
```

zm711, 2024-07-11:
Another random failure in python 3.9 on Windows. From https://github.com/SpikeInterface/spikeinterface/actions/runs/9897996103/job/27343788820?pr=3180
```
FAILED src/spikeinterface/preprocessing/tests/test_silence.py::test_silence - AssertionError: assert 0.1109969 < 0.1
 +  where 0.1109969 = <built-in method sum of numpy.ndarray object at 0x000002B600FB4E70>()
 +    where <built-in method sum of numpy.ndarray object at 0x000002B600FB4E70> = array([0.08974379, 0.02125311], dtype=float32).sum
 +      where array([0.08974379, 0.02125311], dtype=float32) = <ufunc 'absolute'>((array([0.90464187, 0.9732841 ], dtype=float32) - array([0.99438566, 0.99453723], dtype=float32)))
 +        where <ufunc 'absolute'> = np.abs
 +        and   array([0.90464187, 0.9732841 ], dtype=float32) = <function std at 0x000002B64658BBF0>(array([[-0.11603655, -0.73109096],\n       [ 0.8392864 , -0.5543756 ],\n       [ 0.3452874 ,  0.68456554],\n       ...,\n       [-1.1869012 , -0.74847776],\n       [-0.66878515, -0.20611839],\n       [-0.93528956,  1.155699  ]], dtype=float32), axis=0)
 +          where <function std at 0x000002B64658BBF0> = np.std
= 1 failed, 2269 passed, 42 skipped, 749 deselected, 54 warnings in 114.73s (0:01:54) =

```

samuelgarcia, 2024-07-12:
c'est terrible!

zm711, 2024-07-12:
@h-mayorquin, @JoeZiminski , @chrishalcrow,

As briefly mentioned in #3189. We can leave this open and if you see a test failure that appears to be due to a seeding issue/randomly fails put it here. We can track them and make sure that it really is just seeding and not due to a deeper bug when we have more data.

> c'est terrible!

le ci a besoin de ton anniversaire pour fonctionner. C'est fondamental pour le generator.

samuelgarcia, 2024-07-12:
héhé

h-mayorquin, 2024-07-12:
If we set the seed it won't fail. I rather open another issue if somebody wants to improve this test or search for where it was failing. Otherwise, I am afraid this will remain open forever : P