SpikeInterface/spikeinterface #3192: Numpy version within docker container

OlivierPeron, 2024-07-12:
Hello, 

I am using the sorters through dockers, and especially **Kilosort 2.5**, but since **numpy 2** has been released, all my sorting attempts failed. When installing numpy in the container, no version is specified, therefore the latest version is installed, which is not supported by spikeinterface. 
For example, I encountered the following error within the container:

AttributeError: `np.issctype` was removed in the NumPy 2.0 release. Use `issubclass(rep, np.generic)` instead.

Thanks !

zm711, 2024-07-12:
Which version of spikeinterface are you using. I think we have this fixed as of 0.100.8, the 0.101.0rc0, and in main.

OlivierPeron, 2024-07-12:
I am using version 0.101.0rc0 ! I can try to upgrade to new versions !

zm711, 2024-07-12:
You have two options. If you are set on your current version then you can try

```python
run_sorter(..., extra_requirements="numpy==1.26", docker_image=True, ...)
```

Otherwise this should be fixed in 0.101.0 on main (not the release candidate). The release of 0.101.0 on pypi will be soon, but if you want to work on stuff today I would just install from source or force the requirement in the run_sorter function.

OlivierPeron, 2024-07-12:
Ok I will try this ! Looks like the easier way indeed ! Thanks !

OlivierPeron, 2024-07-23:
It worked just fine using your recommendation ! Thanks, I am closing the issue !

closed 2024-07-23 by OlivierPeron