SpikeInterface/spikeinterface #3201: PhySortingExtractor disagreement between docstring and args

bendichter, 2024-07-14:
The docstring says keep_good_only is default `True` but the args are default is `False`

https://github.com/SpikeInterface/spikeinterface/blob/b3383575feade4bf70d623937d6ea2db9cd05b37/src/spikeinterface/extractors/phykilosortextractors.py#L21-L37

zm711, 2024-07-16:
Also same for the unit_locations in postprocessing. The docstring says the default is COM when the actual default is monopolar_triangulation. I'll try to patch both of these soon. Unless you wanted to @bendichter?

alejoe91, 2024-07-16:
I sneaked in a patch in one of the recently merged PRs! 

Fixed on main:
https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/extractors/phykilosortextractors.py#L23

zm711, 2024-07-24:
I'll close this then. The beauty of the numpydoc string for Chris is it should help us start catching all these mismatches.

closed 2024-07-24 by zm711