SpikeInterface/spikeinterface #3202: Out of range float values are not JSON compliant

bxy666666, 2024-07-15:
Traceback (most recent call last):
  File "G:\bxy\spikeinterface-main\examples\get_started\quickstart.py", line 114, in <module>
    w1 = sw.plot_quality_metrics(we_KS2, display=False, backend="sortingview")
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python312\Lib\site-packages\spikeinterface\widgets\quality_metrics.py", line 44, in __init__
    MetricsBaseWidget.__init__(
  File "C:\Python312\Lib\site-packages\spikeinterface\widgets\metrics.py", line 80, in __init__
    BaseWidget.__init__(self, plot_data, backend=backend, **backend_kwargs)
  File "C:\Python312\Lib\site-packages\spikeinterface\widgets\base.py", line 82, in __init__
    self.do_plot()
  File "C:\Python312\Lib\site-packages\spikeinterface\widgets\base.py", line 103, in do_plot
    func(self.data_plot, **self.backend_kwargs)
  File "C:\Python312\Lib\site-packages\spikeinterface\widgets\metrics.py", line 263, in plot_sortingview
    self.url = handle_display_and_url(self, self.view, **backend_kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Python312\Lib\site-packages\spikeinterface\widgets\utils_sortingview.py", line 42, in handle_display_and_url
    url = view.url(label=figlabel)
          ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\bxy\AppData\Roaming\Python\Python312\site-packages\sortingview\views\View.py", line 101, in url
    return fig.url_from_url_dict(self.url_dict(label=label, state=state, allow_float64=allow_float64))
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\bxy\AppData\Roaming\Python\Python312\site-packages\sortingview\views\View.py", line 82, in url_dict
    {"type": view.type, "viewId": view.id, "dataUri": _upload_data_and_return_uri(view.to_dict(), allow_float64=allow_float64)}
                                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\bxy\AppData\Roaming\Python\Python312\site-packages\sortingview\views\View.py", line 153, in _upload_data_and_return_uri
    return kcl.store_json(fig.serialize_data(data, allow_float64=allow_float64), local=local)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\bxy\AppData\Roaming\Python\Python312\site-packages\kachery_cloud\core.py", line 18, in store_json
    text = simplejson.dumps(x, separators=separators, indent=indent, allow_nan=False, sort_keys=True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\bxy\AppData\Roaming\Python\Python312\site-packages\simplejson\__init__.py", line 395, in dumps
    **kw).encode(obj)
          ^^^^^^^^^^^
  File "C:\Users\bxy\AppData\Roaming\Python\Python312\site-packages\simplejson\encoder.py", line 298, in encode
    chunks = self.iterencode(o)
             ^^^^^^^^^^^^^^^^^^
  File "C:\Users\bxy\AppData\Roaming\Python\Python312\site-packages\simplejson\encoder.py", line 379, in iterencode
    return _iterencode(o, 0)
           ^^^^^^^^^^^^^^^^^
ValueError: Out of range float values are not JSON compliant

zm711, 2024-07-15:
@magland, this seems to be a kachery issue. Have you seen this before?

magland, 2024-07-16:
@zm711 I think this probably means that there are NaN values in the dataset, which cannt be serialized to valid JSON. Not sure the best way to remedy that

alejoe91, 2024-07-16:
We should have a remedy in place for that:
https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/widgets/metrics.py#L233-L240

Maybe it's something else?

magland, 2024-07-18:
> We should have a remedy in place for that: https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/widgets/metrics.py#L233-L240
> 
> Maybe it's something else?

It's hard to say... we'd need to somehow intercept that data before it gets to that point and check for things that would trigger

"Out of range float values are not JSON compliant"

zm711, 2024-07-18:
When was that added @alejoe91? Based on the script provided it looks like this is waveform_extractor based code. Maybe that fix wasn't around for this version?

alejoe91, 2024-07-18:
No that has been there for several monts, if not more than a year

zm711, 2024-12-05:
Was this issue ever sorted @bxy666666 ?

zm711, 2025-02-05:
@bxy666666 I will close this as this should be fixed on main and has been fixed for a while if you update your version of spikeinterface. Please reach out if you have continued issues!

closed 2025-02-05 by zm711