SpikeInterface/spikeinterface #3206: Docker for windows server

harshk95, 2024-07-15:
Hi, 
We are just trying to shift to sorting data with SpikeInterface and wanted to run the sorting via machines setup with Windows Server 2022. It appears that Docker does not readily support this and I was wondering if you had any suggesition or pointers for using SpikeInterface on these machines with docker

Thanks a lot!

zm711, 2024-07-15:
Can you do WSL on the servers? I'm not familiar with Windows Servers. But if you could do WSL then you could docker through the linux api and run it that way. We've definitely gotten it working on Windows, but I don't know of anyone on Windows server yet.

JoeZiminski, 2024-07-18:
Hi @harshk95 similar to @zm711 I am not familiar with Windows Server 2022. Is the main proble installing Docker on Windows Server 2022? What kind of errors are you getting? I think if you can successfully setup Docker on Windows Server 2022 e.g. [this guide](https://medium.com/@aravindan27108/installing-docker-on-windows-server-2022-a-step-by-step-guide-0e4ac141095f) then (I'd hope 🤞) SI should run without issue as it is currently working cross-platform when docker / singularity is installed..

harshk95, 2024-07-18:
Hi thanks a lot for the reply and sorry for the sluggish reposnse.
I did follow a similar link, and more or less the same steps and seemed to have docker engine up. Had some problem with WSL though and virtualisation. We are working with the IT at our institute to facilitate the installation on some Windows Server machines we have and hopefully can report how it goes in the coming weeks. I also hope once the docker and WSL2 (which is compatible with Windows Server 2022) are up SI will run smoothly.

zm711, 2024-07-18:
I can also confirm spikeinterface works locally without docker on Windows (10 and 11--again no test on Server). So if that is an option it could be explored as well. But  yeah just let us know. We are more than happy to troubleshoot more and if anything you find out might be useful to future users of the project. :)

zm711, 2024-12-03:
Hey @harshk95, 

any word on this?