SpikeInterface/spikeinterface #3209: Question about Kilosort 2

gabrielstine, 2024-07-15:
Hi Guys,

I am using KS2 in spikeinterface. Every time I run the sorter, SI will save the recording in the sorter output folder. This save seems slow—I assume it does not use parallelization. KS2 will then do whitening and save the temp_wh.dat file, which is also slow. I'm wondering if it's possible to do the preprocessing/whitening in SI, save the preprocessed data to binary using parallelization, and then point KS2 to this file for spikesorting. My sense is this would speed up my pipeline considerably and be more efficient storage wise.

Gabe

zm711, 2024-07-15:
Hey @gabrielstine,

So writing the recording does use parallelization if you have set `n_jobs>1` in your job_kwargs. I tend to prefer using 

```python
si.set_global_job_kwargs(n_jobs=x) # + other kwargs
```

So if you aren't getting parallelization then we need to troubleshoot a bit! As far as your other question  
> I'm wondering if it's possible to do the preprocessing/whitening in SI, save the preprocessed data to binary using parallelization, and then point KS2 to this file for spikesorting

Yes it is. The params are [here](https://github.com/SpikeInterface/spikeinterface/blob/7562b247bd5d3dc2c6e7c8723ab104beefbef1ab/src/spikeinterface/sorters/external/kilosort2.py#L36-L59), but what you want is `skip_kilosort_preprocessing=True`. We have options for [referencing your data](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/preprocessing/common_reference.py)  as well as [whitening](https://github.com/SpikeInterface/spikeinterface/blob/main/src/spikeinterface/preprocessing/whiten.py).

Want to give it a go and if you run into more problems let us know? 

Zach

JoeZiminski, 2024-07-18:
I am not certain on this but I think for file-writing, there is not much of a speed-up for parallelising across CPUs (discussed [here](https://uk.mathworks.com/matlabcentral/answers/567093-is-it-possible-to-accelerate-the-speed-of-saving-data-into-files-with-parallel-way) and [here](https://stackoverflow.com/questions/10254147/writing-data-to-disk-in-parallel)) even with SSD. For reading there are some speedups with SSD but not always HDD (discussed [here](https://pkolaczk.github.io/disk-parallelism/)). I think in this case the only way to speed up is to avoid writing completely. I don't think it is a possibility to avoid writing the `temp_wh.dat` file as KS2 writes this internally (I think after preprocessing before sorting). 

It might be possible to avoid spikeinterface writing the recording to sorting output folder, I'm sure I have seen the `recording.dat` writing skipped but I can't remember the circumstances, I think when the preprocessed data has already been saved to spikeinterface format (so actually, doesn't get around the problem). What is your preprocessing pipeline?

zm711, 2024-12-03:
> , I'm sure I have seen the recording.dat writing skipped

I'm pretty sure the wrapper checks if the file is already binary and int16 and will skip rewriting if pointed. But as Joe said the `temp_wh` is a limitation of KS2 so we can't really help with not writing that. I'll close for now, but open a new issue or reopen if we need more discusssion.

closed 2024-12-03 by zm711

samuelgarcia, 2025-01-06:
> I'm pretty sure the wrapper checks if the file is already binary and int16 and will skip rewriting if pointed. But as Joe said the `temp_wh` is a limitation of KS2 so we can't really help with not writing that. I'll close for now, but open a new issue or reopen if we need more discusssion.

yes it skip if already int16 but it also need need the same channel number, so if some channels are out (like the sync in spikeglx) then a copy is done anyway.