SpikeInterface/spikeinterface #3212: About recording electrodes

ohkoibln, 2024-07-16:
Hello, I would like to know how to identify which electrodes recorded the signal for each unit after obtaining units using Kilosort2?

zm711, 2024-07-16:
@ohkoibln,

we have a variety of ways to get this information by calculating the `unit_locations` with your preferred method. You can see the methods [here](https://github.com/SpikeInterface/spikeinterface/blob/7562b247bd5d3dc2c6e7c8723ab104beefbef1ab/src/spikeinterface/postprocessing/unit_locations.py#L29).

What version of spikeinterface are you using? If you are using from source or the 0.101.0.rcX then you would make a sorting analyzer and then compute to get these values. 

[This tutorial](https://spikeinterface.readthedocs.io/en/stable/how_to/get_started.html) discusses how to calculate various value including locations for versions 0.100.8 and below. [this tutorial](https://spikeinterface.readthedocs.io/en/latest/get_started/quickstart.html#quickstart-tutorial) does the same for the newer api for version 0.101.0.

ohkoibln, 2024-07-17:
Thanks, but I not only need the information of unit_locations ,I also need others. For example, electrodes 47,8759,14778 recorded the signal of unit 3

ohkoibln, 2024-07-17:
> @ohkoibln,
>
> we have a variety of ways to get this information by calculating the `unit_locations` with your preferred method. You can see the methods [...]

Thanks but I already had  the information of unit_locations, I need something else. For example, electrodes 4,7889,24457 recorded the signal of unit 4.

ohkoibln, 2024-07-17:
> > @ohkoibln，
> > 我们有多种方法可以通过计算您喜欢的方法获取此信息。您可以[在此处](https://github.com/SpikeInterface/spikeinterface/blob/7562b247bd5d3dc2c6e7c8723ab104beefbef1ab/src/spikeinterface/postprocessing/unit_locations.py#L29)`unit_locations`查看方法。[](https://github.com/SpikeInterface/spikeinterface/blob/7562b247bd5d3dc2c6e7c8723ab104beefbef1ab/src/spikeinterface/postprocessing/unit_locations.py#L29)
> > 您使用的是哪个版本的 spikeinterface？如果您使用的是源代码或 0.101.0.rcX，那么您可以制作一个排序分析器，然后计算以获取这些值。
> > [本教程](https://spikeinterface.readthedocs.io/en/stable/how_to/get_started.html)讨论了如何计算 0.100.8 及以下版本中的各种值（包括位置）。[本教程](https://spikeinterface.readthedocs.io/en/latest/get_started/quickstart.html#quickstart-tutorial)对 0.101.0 版本的较新 API 执行相同操作。
> 
> 谢谢，但我已经获得了 unit_locations 的信息，我需要其他信息。例如，电极 4,7889,24457 记录了单元 4 的信号。

I want to know more about every unit, Thanks!!

alejoe91, 2024-07-17:
@ohkoibln It's a bit unclear what information you would like to know. Can you be more specific? Do you need a set of channels on which each unit appears?

If that's the case, you can get it from the analyzer sparsity:
```
print(analyzer.sparsity.unit_id_to_channel_ids)
```

ohkoibln, 2024-07-17:
> @ohkoibln不太清楚您想要了解什么信息。您能更具体一点吗？您需要一组每个单元出现的频道吗？
> 
> 如果是的话，你可以从分析器稀疏度中获取它：
> 
> ```
> print(analyzer.sparsity.unit_id_to_channel_ids)
> ```

Gratefully thanks I think that is what I mentioned, Thanks a lot. May I learn more about every unit, Is there any unit guideline?

zm711, 2024-07-18:
As Alessio said I don't think either of us are 100% sure what you are asking for. Do you want something different from what Alessio suggested or are you asking how to get that information for every unit?

bxy666666, 2024-07-19:
> 正如 Alessio 所说，我认为我们俩都不能 100% 确定您的要求。您想要与 Alessio 建议的不同的东西吗，还是您问如何获取每个单元的信息？

Maybe I  have got the information from the other replay concerning features. Thank you again!

zm711, 2024-12-04:
This seems to be solved for the moment, but feel free to open new issues if needed!