SpikeInterface/spikeinterface #3214: Spikeinterface - Spikeinterface-gui interaction

Antho2422, 2024-07-16:
Hi all, 

I was trying to create a tutorial for my team about how to use spikeinterface and I ran into an issue with the spikeinterface-gui module. 

My goal was to use sorting-guy backend to plot the sorting summary as follows :

```python
si.plot_sorting_summary(analyzer, backend='spikeinterface_gui')
``` 

My env add the following versions `spike interface.__version__ = 0.101.0` and `spikeinterface_gui.__version__ = 0.9.0` but both of those versions are not available now. When I updated both packages to the last versions (spikeinterface now in version 0.101.0rc0 and spikeinterface gui in version 0.8.0) I ran into an issue with spikeinterface gui because this was a version without SortingAnalyzer implemented yet. 

How can I manage to be able to use both packages without any issue and including the sorting analyzer ? 

Thanks, 
Cheers, 
Anthony

samuelgarcia, 2024-07-16:
Salut Anthony.
spikeinterface-gui will be released this week.
For now you need to install from source.

Antho2422, 2024-07-17:
Salut Samuel, 
I installed both spikeinterface and spikeinterface-gui from source and I got this error : 

```python
ERROR: Could not find a version that satisfies the requirement spikeinterface>=0.101.0 (from spikeinterface-gui) (from versions: 0.9.1, 0.9.9, 0.10.0, 0.11.0, 0.12.0, 0.13.0, 0.90.0, 0.90.1, 0.91.0, 0.92.0, 0.93.0, 0.94.0, 0.95.0, 0.95.1, 0.96.0, 0.96.1, 0.97.0, 0.97.1, 0.98.0, 0.98.1, 0.98.2, 0.99.0, 0.99.1, 0.100.0, 0.100.1, 0.100.2, 0.100.3, 0.100.4, 0.100.5, 0.100.6, 0.100.7, 0.100.8, 0.101.0rc0, 0.101.0rc1)
ERROR: No matching distribution found for spikeinterface>=0.101.0
``` 

The version 0.101.0 is not available I have the version 0.101.0rc1 right now...

zm711, 2024-07-17:
Hey Anthony, yeah I think that is our bad.

```python
parse('0.101.0rc0') > parse('0.101.0')
Out[19]: False
parse('0.101.0rc0') == parse('0.101.0')
Out[21]: False
```
@samuelgarcia @alejoe91, I think this is part of the problem that @h-mayorquin was talking about no? Where release candidates in this case are always smaller than actual releases so maybe we need to fix the parsing for the gui for the future. 

@Antho2422 the release is coming soon so if you can wait a couple days it should be fine unless we want to change the toml to non rc for now?

Antho2422, 2024-07-17:
@zm711 

Hey ! Thank you, no problem I've discussed with @yger this morning and I'll wait for the release it is not that urgent ! 

Thank you !

h-mayorquin, 2024-07-19:
Yes, that's the problem that I found. I was relying on the release candidate passing that equality but is actually lower.

zm711, 2024-07-26:
@Antho2422, I'll close this with the recent releases, but feel free to open a new issue if needed :)

closed 2024-07-26 by zm711