SpikeInterface/spikeinterface #3216: How to create a channel map for a microwire array electrode?

li1901, 2024-07-17:
I haven't found any information in probeinterface about how to create this type of electrode. My electrode is similar to the  Utah array.
[image]

zm711, 2024-07-18:
Sorry for the delay. Yeah this is definitely a probeinterface issue. But in general you need need to know your geometry which is missing form this diagram and then the kedouBC is attached to what. Is that an adaptor for a headstage?

li1901, 2024-07-22:
> Sorry for the delay. Yeah this is definitely a probeinterface issue. But in general you need need to know your geometry which is missing form this diagram and then the kedouBC is attached to what. Is that an adaptor for a headstage?

Thank you for your response. I currently believe that I can directly use a planar channel map to replace my current electrodes, as I feel that during the sorting process, what affects the results is merely the relative positioning between electrode points, and it has no relation to factors like the angle at which the electrodes are inserted into the brain. Therefore, I think a 2D map can be used directly.

li1901, 2024-07-25:
> Yes that's true, but what I'm saying is we need to know what your recording equipment it. So in this image you have what looks to be the probe and then maybe an adaptor, but what is the adaptor plugged into. Making a channel map requires the last step of mapping the probe to the actual recording equipment (ie the headstage). so how does this interact with the headstage? We have a probe making tutorial [here](https://github.com/SpikeInterface/SpikeInterface-Training-Edinburgh-May24/tree/main/hands_on/probe_handling). But the idea is:
> 
> contact_id -> adaptor (sometimes) -> device_channel_index -> channel_id (for sorting). Right now you have contact_id info but we need to determine the device_channel_index.

Thank you for your reply. The first image below shows my electrode, which integrates an omenstic 32-channel connector. The acquisition equipment I'm using is the Intan RHS 2116 headstage and the Intan RHS Stim/Recording System, as shown in the second images. I understand what you mean by channel_index; I've read through some tutorials and created a usable channel map, and I have now achieved sorting. Thank you for your response.
[image]
[image]