SpikeInterface/spikeinterface #3222: Set subplot parameters in UnitSummaryWidget

florian6973, 2024-07-17:
Hi!

I am currently using the UnitSummaryWidget that I find very practical, however I noticed that we cannot change the parameters of the subplots like the AmplitudesWidget: in the parameters of the AmplitudeWidget we can change the number of bins but here we do not have access. 
[image]

Could we potentially add to the unit summary widget additional fields to pass parameters to subwidgets please? I can make a small PR if needed. Thanks!

Best,

zm711, 2024-07-18:
I think this is a great idea. I've been meaning to suggest something like this for a while.  As long as @samuelgarcia or @alejoe91 don't have a reason not to I would say go for the PR. I would definitely use this!

alejoe91, 2024-07-18:
Yep! Please go ahead with the PR!

zm711, 2024-12-03:
This is all done so I'll close now :)

closed 2024-12-03 by zm711