SpikeInterface/spikeinterface #3223: SI Compatibility with HS2 4.0 Lightning?

litzj01, 2024-07-18:
Hello! Herding Spikes recently released a new updated 4.0 version of their sorter. Is this currently compatible with the latest Spike Interface version? I am using the most recent version of Spike Interface, but when I run `pip install --upgrade herdingspikes`, the HS2 0.3.104 version is installed. Is there another way to upgrade to the 4.0 version of HS2 or is Spike Interface not yet compatible with it?

chrishalcrow, 2024-07-18:
Hello, there's a pull request ongoing here: https://github.com/SpikeInterface/spikeinterface/pull/3210
So you could work off that PR for now. I'm pretty sure this will get merged very soon!

mhhennig, 2024-07-18:
Could you also check you have Python version >=3.12. The new version requires this.
If you use Windows, please use the code from the repo for now: https://github.com/mhhennig/HS2
There's an issue with VC we're trying to solve.

litzj01, 2024-07-18:
Thanks for your reply! I will need to update my python version and then will try again. I am using a Unix based computing cluster, instead of windows at the moment.

litzj01, 2024-07-18:
Hello! I have installed Python 3.12 and was successful in getting HS2 4.0 to install. However, when I ran a sort, I got the error that is mentioned in the pull request #3210 regarding saving the sort: `AttributeError: 'HSClustering' object has no attribute 'expinds'`. Can you explain what the solution was to fix this?

chrishalcrow, 2024-07-19:
For me, I hadn't updated to the latest version of herdingspikes from git. Make sure that's up to date.

mhhennig, 2024-07-20:
This is now fixed. You can use the pip installed version of herdingspikes. Thanks for raising it.

closed 2024-07-20 by mhhennig