SpikeInterface/spikeinterface #3225: About PCA

bxy666666, 2024-07-18:
I want to know the the names of features I put into PCA

alejoe91, 2024-07-18:
There are no names associated to PCA projections! It's just dimension 1, 2, 3 etc

bxy666666, 2024-07-18:
> PCA 投影没有相关名称！只有维度 1、2、3 等

Thanks for your reply， I mean the features extract form the raw data

zm711, 2024-07-18:
It's not that kind of PCA @bxy666666 . This is projecting the waveforms into PC space not projecting all data into PC space. If you working with raw data you likely want to sort the data with a sorter. The sorter itself can internally use PCA, but our PCA metric is for assessing quality of the waveforms. Or you could incorporate into your own spike-sorter if you wanted. I guess.

bxy666666, 2024-07-18:
> 这不是那种 PCA@bxy666666。这是将波形投影到 PC 空间，而不是将所有原始数据投影到 PC 空间。如果您使用原始数据，您可能希望使用分类器对数据进行分类。分类器本身可以在内部使用 PCA，但我们的 PCA 指标用于评估波形的质量。或者，如果您愿意，您可以将其纳入您自己的尖峰分类器。我想。

Thanks !so  there are many features about waveforms?

zm711, 2024-07-18:
yeah the general idea for the PCA is you project the waveform (I believe our default is a few ms worth of data) so you'd have between 50-80 samples. So you would start with those dimensions for the waveform + the channel locations based on the sparsity and then run a PCA on this (you choose number of components and whether to do the PCA globally or locally). So the PC's themselves don't mean anything in reality because they are a mix of samples and channels etc. (so it is bad practice to say PC1 is clearly equal to channel 1 although some people try to make those types of conclusions) It just maximizes the variance/ component but sorting algorithms use PCA because it is one method to get more information with small data structures and quality metrics use them for similar reasons. You can see our docs on the pca-based quality metrics: https://spikeinterface.readthedocs.io/en/stable/modules/qualitymetrics.html

zm711, 2024-07-18:
This is in contrast to your raw data which would have millions of samples on potentially 100s of channels. If you are looking for peak detection we do have functions for that:

https://github.com/SpikeInterface/spikeinterface/blob/c929e514533cb8f4e19dd5f77ade8a4b5f6309f8/src/spikeinterface/sortingcomponents/peak_detection.py#L52

bxy666666, 2024-07-18:
> yeah the general idea for the PCA is you project the waveform (I believe our default is a few ms worth of data) so you'd have between 50-80 samples. So you would start with those dimensions for the [...]

what I want to do is to explain the result as features. For example which feature of data is the most important.

zm711, 2024-07-18:
I think we need to understand what you mean by feature. 

For example the 1st principal component explains the highest variance of the input data. But this 1st principal component is likely actually a mix of different aspects of the recording. Like I said above it could be that sample 22 on channel 1 is the important data and so those two dimensions would become fused together to make the 1st principal component. It is a non-trivial process to try and take principal components and translate them back into features of the raw input data.

bxy666666, 2024-07-18:
> 我认为我们需要了解您所说的“特征”是什么意思。
> 
> 例如，第一个主成分解释了输入数据的最高方差。但这个第一个主成分实际上可能是记录的不同方面的混合。就像我上面说的，通道 1 上的样本 22 可能是重要数据，因此这两个维度将融合在一起形成第一个主成分。尝试提取主成分并将其重新转换为原始输入数据的特征是一个不平凡的过程。

Well, maybe the highest variance of the input data is the feature I mentioned

JoeZiminski, 2024-07-18:
Hi @bxy666666, do you mean instead of perform PCA on the action potential waveforms you want to extract waveform features (e.g. amplitude, rise time, half width, decay time etc) and then run PCA on these?

bxy666666, 2024-07-18:
> 你好@bxy666666，你的意思是，你不想对动作电位波形执行 PCA，而是想提取波形特征（例如幅度、上升时间、半宽、衰减时间等），然后对这些特征运行 PCA？

Yeah，so that it will be explicable!

alejoe91, 2024-07-18:
Then you should take a look at `template_metrics`! https://spikeinterface.readthedocs.io/en/latest/modules/postprocessing.html#template-metrics

bxy666666, 2024-07-18:
> 那么你应该看看`template_metrics`！https://spikeinterface.readthedocs.io/en/latest/modules/postprocessing.html#template-metrics

Thanks a lot！I think this will solve my question.

closed 2024-07-18 by alejoe91