SpikeInterface/spikeinterface #3233: Filtering ConcatenateSegmentRecording on the fly (kilosort4)

carlacodes, 2024-07-19:
I have not encountered this error before -- I am running spikeinterface==0.100.8 and kilosort==4.0.12; I ran kilosort4 via spikeinterface on a list of recordings that were concatenated together on a preprocessing step. I did not save the raw concatenated binary file (so about 200 minutes of recording total).

Is there a way to apply bandpass filtering to a ConcatenateSegmentRecording object on the fly without causing memory issues?

Apologies if this has already been discussed before.
Error below:

```
  File "/nfs/nhome/live/carlag/neuropixels_sort/rat_robot_kilosort4_iblcbin_postprocessing.py", line 145, in <module>
    main()
  File "/nfs/nhome/live/carlag/neuropixels_sort/rat_robot_kilosort4_iblcbin_postprocessing.py", line 141, in main
    pipeline_helpers.spikesorting_postprocessing(sorting, output_folder, datadir)
  File "/nfs/nhome/live/carlag/neuropixels_sort/helpers/pipeline_helpers.py", line 75, in spikesorting_postprocessing
    we = si.extract_waveforms(rec, sorting, outDir / 'waveforms_folder',
  File "/nfs/nhome/live/carlag/.conda/envs/ksenv4/lib/python3.9/site-packages/spikeinterface/core/waveform_extractor.py", line 1657, in extract_waveforms
    sparsity = precompute_sparsity(
  File "/nfs/nhome/live/carlag/.conda/envs/ksenv4/lib/python3.9/site-packages/spikeinterface/core/waveform_extractor.py", line 1785, in precompute_sparsity
    local_we = extract_waveforms(
  File "/nfs/nhome/live/carlag/.conda/envs/ksenv4/lib/python3.9/site-packages/spikeinterface/core/waveform_extractor.py", line 1672, in extract_waveforms
    we = WaveformExtractor.create(
  File "/nfs/nhome/live/carlag/.conda/envs/ksenv4/lib/python3.9/site-packages/spikeinterface/core/waveform_extractor.py", line 320, in create
    return cls(
  File "/nfs/nhome/live/carlag/.conda/envs/ksenv4/lib/python3.9/site-packages/spikeinterface/core/waveform_extractor.py", line 86, in __init__
    self.set_recording(recording, rec_attributes, allow_unfiltered)
  File "/nfs/nhome/live/carlag/.conda/envs/ksenv4/lib/python3.9/site-packages/spikeinterface/core/waveform_extractor.py", line 702, in set_recording
    raise Exception(
Exception: The recording is not filtered, you must filter it using `bandpass_filter()`.If the recording is already filtered, you can also do `recording.annotate(is_filtered=True).
If you trully want to extract unfiltered waveforms, use `allow_unfiltered=True`.
```

chrishalcrow, 2024-07-19:
Hi Carla,

The waveform extractor checks to see if a recording is filtered by looking at `recording.get_annotation('is_filtered')`. It sounds like you've filtered your recording but for some reason this annotation hasn't been updated when you did that. You can set the annotation yourself by taking your recording, let's say it's called `my_recording`, and setting

`my_recording.annotate(is_filtered=True)`

before you use it in your sorting. Then this error should go away.

Another possibility is that you've accidentally not filtered your recording. To check this, could you post some more of your code, including the bit where you do the preprocessing?

Hope that helps!

chrishalcrow, 2024-07-19:
Hi Carla,

Is it working now?

Is that your full `spikeinterface_recording.json` file? Or is there more in it about the preprocessing steps?

It's good that the `multirecordings` knows about the raw rec. It uses the raw recording, and applies the preprocessing steps when passing to kilosort, and it can do this without having to write a big file.

Also I noticed that you're now using spikeinterface version 101. There's a big chance in this new version about what happens _after_ sorting. You'll want to create a sorting analyzer by doing something like:

```
sorting_analyzer = si.create_sorting_analyzer(recording = your_rec, sorting = your_sorting)
```
read more about it here: https://spikeinterface.readthedocs.io/en/latest/modules/core.html#sortinganalyzer

carlacodes, 2024-07-19:
Hi Chris, thanks for your comment -- it was working when I manually set is_filtered = True, like what you said to do:

`my_recording.annotate(is_filtered=True)`

 but I am paranoid that it's not actually using filtered traces. 

No, sorry, I only included a snippet since the spikeinterface_recording.json file is so large, you can see it has metadata about the preprocessing steps here:
```
                "class": "spikeinterface.preprocessing.common_reference.CommonReferenceRecording",
                "module": "spikeinterface",
                "kwargs": {
                    "recording": {
                        "class": "spikeinterface.preprocessing.filter.BandpassFilterRecording",
                        "module": "spikeinterface",
                        "kwargs": {
                            "recording": {
                                "class": "spikeinterface.preprocessing.phase_shift.PhaseShiftRecording",
                                "module": "spikeinterface",
                                "kwargs": {
                                    "recording": {
                                        "class": "spikeinterface.extractors.cbin_ibl.CompressedBinaryIblExtractor",
                                        "module": "spikeinterface",
                                        "kwargs": {
                                            "folder_path": "Y:\\carlag\\robot_maze_data\\raw_data\\RatHC2\\12-07-2024\\RatHC2_12-07-2024_12h12_g13\\RatHC2_12-07-2024_12h12_g13_imec0",
                                            "load_sync_channel": false
                                        },
                                        "version": "0.100.8",
                                        "annotations": {
                                            "is_filtered": false,
                                            "probe_0_planar_contour": [
```
So even though is_filtered = false, it is actually filtering the data when I load:

`sorting = si.read_sorter_folder(output_folder)
`
I am using data compress using mtscomp, so this could be the issue.

I've been switching back and forth from 101 and 100.8 -- should I be using create_sorting_analyzer rather than read_sorter_folder?

chrishalcrow, 2024-07-21:
Ok, great. I'm glad it's working. You will be able to tell if the filtered have been applied or not when you look at your waveforms or templates. Have you been able to generate these and have a look? They will probably look like nonsense if the filtering wasn't applied.

You can also look at your filtered traces by using `plot_traces`, something like
```
plot_traces(my_filtered_recording)
```
should look a lot different than
```
plot_traces(my_raw_recording)
```

Your json file saves metadata about the recording at each step. It's structured something like

```
{
    "class": CommonReferenceRecording,
    "kwargs": {
        "recording": {
            "class": BandpassFilterRecording,
            "kwargs": {
                "recording": {
                     "class": RawRecording
                     "annotations" : { "is_filtered": false }
                 },
             "annotations" : {"is_filtered": true }
              },
         "annotations" : {"is_filtered": true }
    }
}
```
(with LOTS more stuff in it too). So the annotations are saved for each step. For the raw recording, it's correct that it says`"is_filtered": false` since it is the raw recording. But the annotations related to the "CommonReferenceRecording" should have `"is_filtered": true`. If you search through your json file for "is_filtered" you can see if this annotation gets updated as the preprocessing steps are applied. Is it?

You can still use `load_sorting` to load a sorting. If you make and save a sorting_analyzer you'll need to use `load_sorting_analyzer` to load it.

carlacodes, 2024-07-22:
Hi Chris, thanks for your help; yes, I can confirm the steps get updated in the JSON file--I honestly haven't been able to reproduce the error when running it on my local machine, so it's likely something I did where I commented out my preprocessing function, or something funny I did on my cluster environment. I am closing this issue. Thanks so much again for your help.

closed 2024-07-22 by carlacodes