SpikeInterface/spikeinterface #3236: backend="ipywidgets"

410pfeliciano, 2024-07-19:
If I run the following code in SI version 0.100.7, it works fine, but in versions 0.100.8 or 0.101.0rc1, it doesn't.

Here is the code running in the VS Code Jupyter Notebook on a Windows 11 computer:

`%matplotlib widget`
`si.plot_traces(raw_rec, backend="ipywidgets", mode='line', channel_ids=raw_rec.channel_ids[:8])`

This gives the following error:
---> [13](file:///C:/Users/pdrfl/Anaconda3/envs/si_101rc1/Lib/site-packages/spikeinterface/widgets/utils_ipywidgets.py:13) assert "ipympl" in mpl_backend, "To use the 'ipywidgets' backend, you have to set %matplotlib widget"

AssertionError: To use the 'ipywidgets' backend, you have to set %matplotlib widget

The problem is only with `backend="ipywidgets"`, while using other options like `backend="ephyviewer"` works fine.

zm711, 2024-07-19:
In your script did you run `%matplotlib widget` at the beginning?

410pfeliciano, 2024-07-19:
Yes, I use the `%matplotlib widget` at the beginning. In version 0.100.7, it works fine, but now I'm using the latest SI version, and in that, the same code just doesn't work.

zm711, 2024-07-19:
How did you do your new install? (Also we just released 0.101.0 today if you want to upgrade to that--but there is one big API change there ).

410pfeliciano, 2024-07-19:
I did the installation using the `full_spikeinterface_environment_rolling_updates.yml`. I'll try to install the new release.

410pfeliciano, 2024-07-19:
I installed the new version. If I run the `print(f"SpikeInterface version: {si.__version__}")` the Output is: `SpikeInterface version: 0.101.0`

But still, I have the same error. The installation was done using the `full_spikeinterface_environment_windows.yml` file, which has the [full, widgets] specification.

zm711, 2024-07-19:
And where and how are you running your script? I have tried install from source with the widget and it worked a month ago..

410pfeliciano, 2024-07-19:
Right now, I'm running the script in VS Code Jupyter notebook. But I also ran it in Jupyter notebook alone too and had the same problem. Interestingly, I don't think the problem is Jupyter or the ipywidgets installation since I have tried some basic tutorials using the ipywidgets library and they seem to work fine.

h-mayorquin, 2024-07-19:
I confirm that I can reproduce the error:

```python
%matplotlib widget

from spikeinterface.core import generate_recording

recording = generate_recording(durations=[10.0])
from spikeinterface.widgets import plot_traces

plot_traces(recording, backend="ipywidgets", mode='line', channel_ids=recording.channel_ids[:8])
```

h-mayorquin, 2024-07-19:
Fix in https://github.com/SpikeInterface/spikeinterface/pull/3238

If any of you two guys want to keep the torch on that one and figure out what's going on it would be great.

The patch is working though.

410pfeliciano, 2024-07-19:
Changing `%matplotlib widget` to `%matplotlib ipympl`, as suggested by h-mayorquin, worked.

closed 2024-07-21 by 410pfeliciano

chrishalcrow, 2024-07-21:
I can reproduce the error on Linux, but not on Mac. With
```
ipympl 0.9.4
ipywidgets 8.1.3
matplotlib 3.9.0
```
on both.

taningh86, 2024-07-31:
Hi. Any solution to this issue yet? I am having the same issue with version 101 and %matplotlib ipympl also don't work. I was about to write an issue about this but found this issue on time.

h-mayorquin, 2024-07-31:
Should be fixed on the main branch.

taningh86, 2024-07-31:
@h-mayorquin my version of fixing it was removing the environment, redownloading the full_spikeinterface_environment_windows.yml file, creating the environment with that (instead of using pip), and running check_your_install.py and cleanup_for_windows.py. Then using "%matplotlib ipympl".

h-mayorquin, 2024-07-31:
Mmm if you use the version from github just pip install . -U in the pertinent environment should work. But I am glad you solved your problem!